iizukanao/picam #12: How do you think, is it possible to create python bindings to picam?

yurket, 2015-01-19:
It would be great to have some more high-level way to start\stop recording. So I consider to create python bindings for picam, but I've never done it before. So maybe you can give me some directions how to do it?
Thanks!

iizukanao, 2015-01-26:
Hi, sorry for the delay.
If you want to just start/stop recording from Python script, create empty start_record and stop_record files in hooks directory.

If you want to control picam directly, it needs some work on picam codebase to make it accessible from outside. Also, I don't know how to create Python binding for C but it must be possible.

yurket, 2015-01-27:
Thanks for your reply!
For now creating files start_record\stop_record is enough, but I may need to control picam directly in the future.

closed 2015-01-27 by yurket

iizukanao, 2015-01-27:
OK, I will create picam API.
Thanks for the suggestion!

rpastorvargas, 2016-04-28:
Any advances on picam python API?

Regards,
rafael

iizukanao, 2016-04-29:
@rpastorvargas Not yet.

rpastorvargas, 2016-05-02:
Ok, thanks for the information. Maybe you can request help on this, probably i'll could help you with some issues and implementation.

regards,
rafael