iizukanao/picam #16: help with picam, node-rtsp-rtmp-server

sheffler, 2015-05-14:
First.  Thank you for your contributions.

Hello - i am having a little trouble understanding how to view the picam.  I installed it successfully and have used "picam --tcpout URL" with "ffplay -i tcp://192.168.1.117:8181?listen" and could view the stream.  Now I am attempting to serve the sstream via node-rtsp-rtmp-server.

This is on an RPi B.

I ran server like this, using the default settings.
% sudo coffee server.coffee

I verify that the server is running by visiting http://192.168.1.81 and I see the "Hello World"

Now, I start Picam like this:
% picam --rtspout

But I am unable to figure out how to view the stream.  I want to try something like
% ffplay -i rtsp://192.168.1.81/live

where that URL is my RPi.  Do you have any suggestion?  How should I be able to view the stream?

Best regards.
Tom

sheffler, 2015-05-14:
I just answered my own question.  Here is where I found the stream.
% ffplay -i rtsp://192.168.1.81:80/live/picam

nice software.

iizukanao, 2015-05-15:
Yes, that's it!

closed 2015-05-15 by iizukanao