iizukanao/picam #17: Motion detection?

spikedrba, 2015-05-15:
Hi there,

first of all fantastic work, really excited to have discovered this project. I was wondering if you had any plans to support motion detection. I realize this might be out of scope, but thought I'd ask since it's something I'm looking into. Also I thought I'd mention this project, which is doing something similar and has motion implemented (uses openmax too):

https://github.com/dickontoo/omxmotion/blob/master/motion.c

best,

Spike

iizukanao, 2015-05-15:
I actually implemented basic motion detection using OpenCV with picam in the past, but that part has been removed from public source code. Look at cam_fill_buffer_done() function in stream.c. There last_video_buffer is set, and it is the YUV420p data you want to use.

If you are going to use OpenCV for motion detection, I think I can provide some part of the source code.

spikedrba, 2015-05-15:
thanks for the quick reply. I was considering using OpenCV especially for face recognition, however it seems that anything done on the CPU side is going to be real slow/low fps, primarily due to working with pixels. A better approach seems to be using motion vectors provided by the x264 encoding proecss, however those aren't as accurate and miss a lot of information. I was pondering combining the two approaches - use the vectors to figure out when there's likely motion and then pass a couple frames to openCV for further analysis without blocking the encoding/streaming.

any thoughts on that?

iizukanao, 2015-05-16:
You don't have to parse every frame with OpenCV. Just parse one frame every two seconds or so. If your image processing task is too heavy for just one frame (e.g. hangs Pi for a few seconds, or slows down FPS from 30 to below 20), you should reduce the resolution of video or do less in image processing. Because video encoding is mostly done by hardware, FPS will not be heavily affected by CPU load unless it is huge.

BTW, I don't know how to extract motion vectors from H.264 bitstream. In stream.c, the first argument of send_video_frame() is the raw H.264 bitstream. Can you extract motion vectors from this?

spikedrba, 2015-05-16:
right, that was the plan, to parse only every I-Frame or something in a GOP, but that's not good enough for motion detection, that needs to happen across all frames and therefore on the GPU through the motion vectors.

Regarding extracting the motion vectors, I'm not really sure myself and still trying to find out. What I've seen from other codebases, both mmal tools and omax ones is that they set a flag like this on the encoder object:

    OERRw(OMX_SetParameter(enc, OMX_IndexParamBrcmVideoAVCInlineVectorsEnable, obool));

and subsequently they are able to read the vectors from the buffer returned by the encoder so I'm guessing they are packed together somehow and you can extract them.

I think this is how Dick does it in omxmotion:
https://github.com/dickontoo/omxmotion/blob/master/omxmotion.c#L1027

iizukanao, 2015-05-17:
OK, I've pushed a new branch "mvec". It writes motion vector data (likely) to a file on each keyframe. To write all motion vector data, set `WRITE_ALL_MVEC_FRAMES` constant to 1 in stream.c. Try it out and see if you can do something with that data.

spikedrba, 2015-05-18:
@iizukanao thanks, will check it out as soon as I can. For my own clarity and for posterity (in case someone comes across this thread), a couple people got back to me and confirmed the following: the code is the same and the encoder will "sometimes" return motion vector frames and others the actual video frames.

I'm yet not clarify what that "sometimes" is, if it's every other frame in a 1:1 ratio or what, but both MMAL and OMAX implementations I've looked at do the same thing: in the piece of code that deals with the buffer returned from the encoder they check the buffer's flags for OMX_BUFFERFLAG_CODECSIDEINFO or MMAL_BUFFER_HEADER_FLAG_CODECSIDEINFO. If that's true than you're getting motion vectors, otherwise it's video data.

hope this helps somebody and I'll get back to you as soon as I have a chance to test this out.

thanks,

iizukanao, 2015-05-18:
I get a motion vector data after each encoded video frame. So, if it's 30fps, I get 30 frames of motion vector data per second. I'm using OpenMAX and apparently it's working well.

You are right that OMX_BUFFERFLAG_CODECSIDEINFO denotes motion vector data. When you get a buffer with that flag, you have to fetch a buffer one more time to get the next video frame.

spikedrba, 2015-05-20:
thanks a bunch for getting this going, I'll jump on it as soon as I resolve my other issues, right now I can't get smooth playback (see other issue I opened if you have a chance).

thanks,

iizukanao, 2015-11-23:
@spikedrba I'd like to know if you have any results on this, as I'm considering whether to merge this code into master.

spikedrba, 2015-11-23:
@iizukanao around the time I started looking into this another project came out that did all of it already so I never pursued any further. Here it is: https://github.com/billw2/pikrellcam

Bill had done an amazing job in working with h264 vectors, he's added all kinds of things to work with noisy and do high quality motion detection at 30fps. You can see some of the action in a video here: http://billw2.github.io/pikrellcam/pikrellcam.html

it's really no good for audio/video streaming, but the security camera part of the project it's perfect.

iizukanao, 2015-11-24:
@spikedrba OK, thanks for letting me know. Closing.

closed 2015-11-24 by iizukanao