iizukanao/picam #18: high lag and blocky video

spikedrba, 2015-05-20:
Hi there,

thanks for all the great work you put into picam and for making it avaliable, it's an incredible headstart for a project I'm working on.

I'm using a p2 (not overclocked) and a linux desktop running ubuntu 14.04 with 8GB ram/3GHz CPU. I'm using the latest binary as per you setup guidelines and starting picam like this:

    ./picam --noaudio --vflip -o /run/shm/hls/

for a quick test I then got apache2 going and put a symlink in /var/www pointing to the hls directory. I opened vlc and played a network stream http://IP_ADDR_PI2/hls/index.m3u8

Playback started immediately but there was a visible gap of several seconds. I put a phone with a timer in front of the camera and observed ~4s delay, which is much higher than your results. On top of that image was really blocky and at the beginning of the playback the first 3/4 rows of the video were garbled.

Just to make sure it's not something odd with my network or camera (it shouldn't, all ethernet going to the same switch) I ran the classic netcat + mplayer test and results were much better: no observable delay and image wasn't blocky.

The blocky/bad video quality is also observable in recorded videos (triggered via the hook stuff as per guidelines).

Am I doing something wrong? with vlc I guess I might be hitting some buffering issues, but I've not had those issues playing other streams. As a side note, I tried to play the stream with my iphone and while it starts fine it stops very quickly as if it wasn't refreshing the playlist but just playing what it originally downloaded.

I will try to go through the manual install to get ffmpeg going so that I can try the rtmp stuff hopefully tomorrow and once I figured out what's wrong and videoplayback is smooth I'll jump back on the motion vector stuff.

thanks again,

Spike

iizukanao, 2015-05-20:
You cannot expect a latency less than 3-4 seconds on HLS. That latency stems from the design of HLS, and we cannot do much about it. For sub-second latency, you should use RTSP or RTMP instead.

To increase video quality, specify a higher video bitrate using -v option (e.g. -v 5000000). It defaults to 2Mbps which is relatively low for 720p at 30fps.

If the recorded video has some weird noise like flashing band, maybe your power supply is insufficient. Use a 2A power supply.

spikedrba, 2015-05-20:
thanks, couple things.

I wasn't aware of the design issue with hls, I'll try to get rtmp going. As to blockiness and flashing:
- I forgot to mention the raspivid command, but I did specify bitrate in it to the same as picam defaults to, so the test should have been comparable. I also tested picam with a high h264 profile, same results
- I'm aware of power problems and I'm already using a 2A power supply, plus if it was such a fundamental issue wouldn't you expect it to surface with raspivid too?

I will test again with higher bitrate and then try to get the rtmp going.

thanks,

iizukanao, 2015-05-20:
Can you provide an example video file with that noise? Without it, I can't further troubleshoot this issue.
Also, please list all of the peripherals connected to your Raspberry Pi.

spikedrba, 2015-05-20:
really puzzled. So.

- the banding is gone, I cannot repro no matter what I try.
- the problem with blockiness is weird, and might depend on something else:
```
PI: raspivid -o - -t 0 -w 1280 -h 720 -vf -fps 30 -b 2000000 | nc 10.11.11.138 5001
Desktop: nc -l 5001 | mplayer -fps 31 -cache 1024 -
```
those parameters should be all the same as picam's default. With that test I see a pretty clear image popping up on my desktop screen. However as I was trying to record a sample for you I did this:
```
raspivid -o ~/test2Mbitrate.h264 -t 10000 -w 1280 -h 720 -vf -fps 30 -b 2000000
```
which is the same command except that it writes to that file instead of stdout. I then scp'ed it over to my computer and played it with mplayer. Lo and behold it was blocky! it looked exactly like the picam's video recording.

I'm not really sure what to make of this, but I guess it's not picam as I originally thought since it looks the same as raspivid when it's a locally saved video file. Blockiness goes away as soon as I increase bitrate as you suggested.

Why the stream with netcat looks better remains a mistery.

closing the issue and sorry for the false alarm.

closed 2015-05-20 by spikedrba