iizukanao/picam #20: Segmentation fault when starting to record

Linkaan, 2015-06-15:
Hi, I have successfully run picam many times before without any errors, but this time when I tried to record from my raspberry pi camera, picam crashed. Maybe it is hardware related? Anyhow here is some debug info retrieved using gdb:

Linux raspberrypi 3.18.6+ #753 PREEMPT Sun Feb 8 14:47:22 GMT 2015 armv6l
(gdb) handle SIGILL nostop
Signal        Stop      Print   Pass to program Description
SIGILL        No        Yes     Yes             Illegal instruction
(gdb) r
Starting program: /home/pi/Fagelmatare-SW/PIRDetect/bin/picam
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".

Program received signal SIGILL, Illegal instruction.

Program received signal SIGILL, Illegal instruction.
[New Thread 0xb6a33470 (LWP 2921)]
configuring devices
[New Thread 0xb6233470 (LWP 2922)]
[New Thread 0xb5a33470 (LWP 2923)]
[New Thread 0xb5233470 (LWP 2924)]
[New Thread 0xb4a33470 (LWP 2925)]
[New Thread 0xb4233470 (LWP 2926)]
[New Thread 0xb3a33470 (LWP 2927)]
disk_usage=76% [New Thread 0xb2f2b470 (LWP 2929)]
start rec to rec/tmp/2015-06-15_22-24-48.ts

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xb2f2b470 (LWP 2929)]
0x000a5aa0 in rec_thread_start () at stream.c:754
754     stream.c: No such file or directory.
(gdb) backtrace
#0  0x000a5aa0 in rec_thread_start () at stream.c:754
#1  0xb6f0fc00 in start_thread () from /lib/arm-linux-gnueabihf/libpthread.so.0
#2  0xb6b00348 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
#3  0xb6b00348 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)

NOTE: I cross compiled picam from the github repository 2015-03-05

Linkaan, 2015-06-16:
Nevermind, it was hardware related, however should that really crash picam?

closed 2015-06-16 by Linkaan

iizukanao, 2015-06-19:
I have no idea how a hardware problem could cause that crash. Can you elaborate on what kind of hardware issue you were having?

Linkaan, 2015-06-19:
It seems to be working just fine after I plugged the camera out and then
put it back. It didn't work with  raspivid either. It gave me a ~20 KB file
with raspivid.

>
> I have no idea how a hardware problem could cause that crash. Can you
elaborate on what kind of hardware issue you were having?

Linkaan, 2015-06-20:
Sorry, I meant to say the file was about 30 bytes big.
 Den 19 jun 2015 12:10 skrev "Linus S" <linus122xbb@gmail.com>:

iizukanao, 2015-06-20:
If the camera module is not properly attached, the following error message should appear.

    $ ./picam
    configuring devices
    error: failed to create camera component: 0xffffffff
    Have you enabled camera via raspi-config or /boot/config.txt?

A camera module is relatively susceptible to electromagnetic interference, and sometimes you will need to reboot the Pi to fix the situation. If you have the steps to reproduce the bug, please let me know.