iizukanao/picam #22: Timestamps in video and sometimes video repeats itself

Linkaan, 2015-07-10:
Hi Nao!
I've been using picam for a while and I've always appreciated the ease of use. There is only one thing that would make picam for me so much better, timestamps. Yes, I'd love it if one could choose to put a timestamp in the corner of the video because that would be really useful for me (and probably many others) considering I'm using a PIR sensor to trigger the camera, and I could perhaps use ffmpeg to add the filename in post-processing, but it would be ideal if you could add timestamps to picam natively! Speaking of PIR sensors, I've noticed that sometimes when stopping and starting to record, picam will sort of overflow the last segment into the next video, it seems as it writes a buffered segment into the next video making the effect when I merge together all videos that the video repeats itself. I don't know if this is because picam doesn't register the stop_record file or maybe it is beacuse I'm using an old version of picam. Anyhow I'd like to thank you for putting all the astonishing amount of effort into picam.

/Sincerely Linus

Linkaan, 2015-07-17:
I think I might have an idea why I experienced the issue of videos overlapping each other, it might be because of the recordbuf option, which will start recording 5 keyframes (default) previous to that I started recording. But for the most part I will probably not bother having that overlap, I'm just a bit curious :)

Linkaan, 2015-08-08:
I implemented a basic timestamp feature based on <a href="https://github.com/Rutong/picam">this</a> fork of picam with the latest code, however on the pi this is terribly inefficient, because the CPU has a hard time keeping up. Perhaps using the cairo library would yield better results as demonstrated in <a href="https://github.com/tasanakorn/rpi-mmal-demo/tree/develop">this</a> github repository. I'd be grateful if you have any suggestions :)

closed 2015-11-21 by Linkaan

iizukanao, 2015-11-23:
@Linkaan Thanks for testing out the new features! Timestamp and subtitle are now available in version 1.4.0. However, overlaying text in realtime is somewhat too heavy for Raspberry Pi 1.