iizukanao/picam #23: Record Maximum File Size

danielgatley, 2015-07-10:
Hi, 

Would it be possible to have a maximum file size for the recorded content (archive), so that when the size is reached a new file (with new timestamp filename) is created?

Oh and thanks for picam, excellent work.

Regards

Daniel

Linkaan, 2015-08-08:
You could create a simple bash script running in the background which would test for the file size of the file and then simply call `touch hooks/stop_record` and `touch hooks/start_record`: See <a href="https://gist.github.com/Linkaan/b42032c38a0ceb49fdc1">this</a> gist.

danielgatley, 2015-08-14:
Thank you for the update, I did end up just writing a simple bash script but appreciate you taking the time to respond.