iizukanao/picam #24: Hardware issue

gregoryking, 2015-08-03:
Hi,

This is not an issue with your tools (which work very well), but I thought given your advanced use of raspberry pis you would be well placed to comment.  Like many, I'm experiencing significant noise on the microphone input which is effectively making my application unusable.  [Posting](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=116451&p=797789) over on raspberry pi forum, someone suggested it was ground loop noise but I'm all out of ideas how to resolve (I've read many posts!)

Let me know if you have any good ideas from your experiences and thanks for your awesome work.

Greg

Linkaan, 2015-08-03:
It might be because you're using a switched-mode power supply with a low enough operating frequency for you to hear it. Have you tried using another power supply with higher quality or with <a href="https://en.wikipedia.org/wiki/Ferrite_bead">ferrite bead</a> which might be able to filter out noise coming from the power supply. However, listening to the sound you posted, I agree that it sounds like noise generated by ground loop. Is the same noise generated if you take your raspberry pi outside? As a last resort you can try some of <a href="https://en.wikipedia.org/wiki/Ground_loop_%28electricity%29#Solutions">these</a> suggestions.

gregoryking, 2015-08-03:
I'm using [http://www.amazon.co.uk/dp/B00JJCS49E/ref=pe_385721_37038051_TE_3p_dp_1](this)  power supply and have also tried a genuine iPad charger which I presume is good quality as well as powering directly from my macbooks USB port - all are noisy though the macbook and power supply have different noise issues. I've updated the post over on the raspberry pi forum with further audio samples.  I've tried powering the pi via a USB adapter in my car (which delivers 2.1A) and the same issues arise.

I might give a ferrite bead a shot as a last resort.

gregoryking, 2015-08-18:
Tried a ferrite bead, didn't make any difference.

Here's a different way of trying to solve the problem, can anyone whose having success with picam whose getting Hugh quality audio along with their video reply with the details of there setup, ie pi version, sound card, OS, power supply etc. I'm keen to try and replicate a setup that works.

Linkaan, 2015-08-18:
I have similar issues but with USB noise, using a capacitor over the USB
pins did help me, you could try with different ones. 2200 uF / 16 v for
starters.

gregoryking, 2015-08-18:
Thanks for your reply, could you advise specifically which USB pins (power supply, sound card?)

Linkaan, 2015-08-18:
Have a look at my topic over on the raspberry pi forums:
https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=117318

iizukanao, 2015-10-28:
Sorry for the late reply. I'm not an expert in this area, but some types of Wi-Fi dongle are prone to pick up noise from Wi-Fi signals. Maybe a different Wi-Fi dongle will help. I'm using [I-O DATA WN-G150UMK](http://www.amazon.co.jp/gp/switch-language/product/B007P9IX8A/ref=dp_change_lang?ie=UTF8&language=en_JP).

gregoryking, 2015-10-28:
Thanks. I actually had some success by changing the cheap sound card to a
slightly more expensive one and using a higher quality microphone. There
was still a fair bit of noise though which I tracked down to the wifi
adapter, especially prominent when the sound card is immediately next to
it. I've found that using a short USB extension on the sound card to get it
away from the wifi adapter (mine is edimax) and placing a ferrite bead on
the USB cable reduces noise significantly.

iizukanao, 2015-10-28:
That's good to hear! Thanks for letting me know.

closed 2015-10-28 by iizukanao