iizukanao/picam #26: API?

snoop244, 2015-09-08:
First of all, thank you for this great application.  I'm using the recordbuf to pre-record segments of video that I can then record to disk if an event worth recording happened in the previous X seconds.  

Some of the things I could use an API for:
-While hooks seem to work for manual execution, it would be great to have start and end record happen through an API (allowing the file name and directory to be set at start and some elements of standard meta data returned from the file at end capture)
-Changing the amount of delayed video captured from the recordbuf would be nice too, but that would probably be hard to implement (e.g. record with 10 seconds pre-video vs 5 seconds). 

Thanks,
Steve

iizukanao, 2015-11-23:
@snoop244 Thanks for the suggestion. Some of the features you requested are now available in the latest version (1.4.0).

> -While hooks seem to work for manual execution, it would be great to have start and end record happen through an API (allowing the file name and directory to be set at start and some elements of standard meta data returned from the file at end capture)

I assume hooks/start_record and hooks/stop_record are already considered API. With any programming language, you can start or stop recording by just creating an empty file. I think it is much more convenient for casual users than messing up with shared libraries or Python binding. In the latest version, you can specify the directory and filename as parameters to hooks/start_record.

    # Start recording to /tmp/myout.ts
    $ echo -e "dir=/tmp\nfilename=myout.ts" > hooks/start_record

Also, some stats are available in state/*.ts file after a recording is complete.

    $ cat state/2015-11-19_01-18-09.ts
    duration_pts=2083530
    duration_sec=23.150333
 
Is it enough for you?

> -Changing the amount of delayed video captured from the recordbuf would be nice too, but that would probably be hard to implement (e.g. record with 10 seconds pre-video vs 5 seconds).

Now you can do that. I have introduced the concept of **global** and **per-recording** recordbuf, and you can use any number of recordbuf for each recording within the limit of global (maximum) recordbuf. For details, see https://github.com/iizukanao/picam#global-and-per-recording-recordbuf

snoop244, 2015-12-06:
This is great news.  Thank you!