iizukanao/picam #27: record bitrate and stream bitrate

timdegouss, 2015-10-13:
Hi, 
First thank you for your tool. I successfully  install Picam and node-rtsp-rtmp-server. I'm using --rtspout flag to enable  output for node-rtsp-rtmp-server in picam  and it is working well. I receive video stream in VLC by opening a network stream with this adress rtsp://PI_HOST:80/live/picam

I want to know if it possible to have one bitrate used by picam to record on pi disk and an other bitrate for streaming ? In my case streaming is only for a preview use so a low bitrate, (even low resolution) for low latency stream is prefered. But i want to keep the best quality for the the video recording on pi.
Is node-rtsp-rtmp-server reencoding video  ? Do I have to change some parameters in  coffee.config ?

thank you !

Linkaan, 2015-10-13:
Hi @timdegouss I had a very similiar issue only I wanted to stream to ustream.tv, you cannot easily do this because you would have to transcode the video using a software encoder which would be too CPU-intensive for the raspberry pi itself. Instead you can transcode the video on another more powerful computer on your local network and stream that over rtsp. If you can use hardware accelerated encoding it might be possible but I am not sure at all, @iizukanao will have to confirm what I said though.

iizukanao, 2015-10-14:
@Linkaan is right. To produce different bitrates, you have to re-encode the video in some other way, since the hardware encoder is occupied by picam.

timdegouss, 2015-10-15:
Thank you @Linkaan and @iizukanao for your answers. I will try to find a way to handle this.

timdegouss, 2015-10-15:
I have an other question, maybe @Linkaan and @iizukanao could tell me if is possible or just crazy, It is technically possible to set preview stream  from camera (CAMERA_PREVIEW_PORT) to an output for node-rtsp-rtmp-server (or something else)?

iizukanao, 2015-10-15:
The camera component only produces YUV420 (raw data), not H.264 (compressed). The video_encode component converts YUV420 into H.264. camera and video_encode are two separate components, and there is only one video_encode component available. So it doesn't make sense to use the preview port in this case.

timdegouss, 2015-10-16:
Ok @iizukanao thank you for clarifying that !

closed 2015-10-28 by iizukanao