iizukanao/picam #28: Recording is sometimes not started

Linkaan, 2015-11-05:
Hi Nao,

Great to see this project is still under active development. I am still working on my bird feeder and if it wasn't for picam I wouldn't be where I am with the project today. I use a timerfd on my application and if the timer of five seconds runs out and the recording will be stopped. This works great for the most part however on some occasions the timer runs out and the recording is stopped and in a blink of an eye later there is motion pursuant to my pir sensor which will try to start the recording but fails because picam didn't have time to cleanup after the last recording.

```
stop rec
disk_usage=41% start rec to rec/tmp/2015-11-05_22-04-03.ts
stop rec
recording is already started
disk_usage=41% start rec to rec/tmp/2015-11-05_22-04-33.ts
stop rec
disk_usage=41% start rec to rec/tmp/2015-11-05_22-04-57.ts
stop rec
recording is already started
```
I looked at the source code and I saw you print out `stop rec` and then write some buffer to `recording_archive_filepath` and this doesn't happen in an instant so during that time my program tries to be a hassle and throw a new request to start a recording but because the `is_recording` flag is still not false picam will exit out early and the recording is never started.

An easy solution on my end would be to have a "cooldown" value whcih will essentialy wait a few seconds after the recording has stopped and only then send a request to picam. Another solution would be for picam to use a timed lock on a common mutex which will block until the recording has stopped or a timeout has passed and only then try to record, but I can see this can be problematic since if somebody blindly tries to start a recording without keeping a local flag and picam is recording already, picam may wait unnecessarily for the recording to "stop". 

A more complex approach would be to somehow after the recording has been stopped and a new request to start a recording has been made within *x* seconds after the recording was stopped, resume recording in the same file and write the buffered data into the file. Does that make sense?

Anyway that's just my frame of mind, please let me know if you have got any ideas!

iizukanao, 2015-11-18:
Thanks for the suggestion. I've just addressed this issue at the commit 8fb85672a077fef6fb49f1ca1d4264240ae7d8a3. Watch for changes on state/record. The contents of state/record changes to "false" when the recording is completed. You can safely start another recording when state/record is false.

Linkaan, 2015-11-19:
@iizukanao Great! I'll keep an eye on the state/record file instead of storing the state of the recording on my side. By the way I couldn't be more grateful that you put the effort in to have subtitles and timestamps, it will really be helpful. I will use it to display the temperature of the CPU, outside temperature and maybe the CPU percentage. You have my thanks.

closed 2015-11-21 by Linkaan