iizukanao/picam #29: latency issues

alisozkesen, 2015-11-14:
Hi,

I've downloaded pre installed version and after a little bit of struggle, it's up and running via nginx. 
That's a great work thank you. 

When I stream via nginx (not the m3u8) (I use VLC on mac) there is a up to 10 seconds latency despite it's working on the local network. Sometimes the sound and the video both freezes and then continues. 

I think "Create symbolic links (optional, but strongly recommended)" may create a observable difference. But as I'm a newbie on linux platforms, I don't know what to do exactly. 

If you don't mind about to write this steps, I would gladly test the situation. 

I've tested a series of resolutions 640x480, via cable lan, overclocking the raspberry but no hope. 
I've been using raspi 2 btw.

thank you
best.
Ali

Linkaan, 2015-11-18:
I don't know if it is supported in picam, but you should be able to change to UDP by specyfing `--tcpout udp://127.0.0.1:8181`. It might help.

iizukanao, 2015-11-18:
I think this is the same problem as #19. VLC is good for RTSP, but not good for RTMP. Use [Strobe Media Playback](http://sourceforge.net/adobe/smp/home/Strobe%20Media%20Playback/) as a client and the latency will be greatly reduced.