iizukanao/picam #31: How to get this to start when Raspberry Pi starts

FWCJohn, 2015-12-29:
I want the camera to start streaming as soon as its plugged in, so I have tried to add a init script to run at startup. For some reason, I can not get picam to run at startup. The nginx server starts up, and for some reason my picam script will not.

Is there any reason Picam can't start from init.d? 

#! /bin/sh
# /etc/init.d/picam

### BEGIN INIT INFO
# Provides: picam
# Required-Start: $remote_fs $syslog $all
# Required-Stop: $remote_fs $syslog
# Default-Start: 2 3 4 5
# Default-Stop: 0 1 6
# Short-Description: picam
# Description: picam
### END INIT INFO

case "$1" in
    start)
        echo "Starting picam"
        sudo mkdir /run/shm/hooks
        sudo mkdir /run/shm/state
        sudo mkdir /run/shm/rec
        sudo /usr/src/picam/picam --tcpout tcp://127.0.0.1:8181
    ;;
    stop)
        echo "Stopping picam"
        CAM_PID=`ps auxwww | grep picam.py | head -1 | awk '{print $2}'`
        kill -9 $CAM_PID
    ;;
    *)
        echo "Usage: /etc/init.d/picam {start|stop}"
        exit 1
    ;;
esac
exit 0

Linkaan, 2015-12-29:
Here's a few things you can try. Did you run `chmod +x /etc/init.d/picam`? Does it work if you manually start the script using `invoke-rc.d picam start` and also since picam requires networking it would be a good idea to create a if-up.d script in `/etc/network/if-up.d` and call the init.d script from there instead. [Here](https://gist.github.com/Linkaan/0d7c3f95f6e7644c2bf0) is the script I am using. Don't forget to run `chmod +x /etc/network/if-up.d/picam`. Don't forget to disable the init.d script from running on boot: `update-rc.d -f picam remove`.

If you can't manually start the script with `invoke-rc.d picam start` I can provide the init.d script I am using.

iizukanao, 2015-12-30:
@FWCJohn Did you try https://github.com/iizukanao/picam/blob/master/etc/init.d/picam ?

FWCJohn, 2015-12-31:
@iizukanao I installed the one you wrote as well, and I can't seem to even get that to start from commandline. It doesnt give an error, but is not running either. 
@Linkaan I also tried your way, and while this seems like its probably pretty good, the pi would hang at LSB interface taking to long. When I remove the if-up.d/picam script, it operates fine (just not booting picam at load)
with my script I can get picam running at commandline.

iizukanao, 2016-01-01:
I think I need to make picam able to daemonize itself.

Linkaan, 2016-01-01:
@FWCJohn It is probably a good idea to try on another raspberry pi or atleast with a fresh install of raspbian to see if your script works correctly. Also I assume with your previous method without using `if-up.d/picam` you have installed your script to start on boot using `sudo update-rc.d picam defaults` (to create the necessary symlinks to run on boot)?

FWCJohn, 2016-01-07:
@Linkaan Thanks I have added that as well. Its almost as if Picam wont start because a program (or maybe network)  it is waiting on hasn't booted yet, but that is just a rudimentary guess. I can even see it attempting to start in the logs, but then it isn't started when I check.

FWCJohn, 2016-01-07:
When I do service -status -all I get a ? next to Picam. Not sure why that isnt a + or -

FWCJohn, 2016-01-07:
@iizukanao I keep trying your bootup script, and for whatever reason I can never get an error to produce. The only daemon log is 

raspberrypi systemd[1]: Started LSB: picam.

but never boots picam even from commandline.

FWCJohn, 2016-01-08:
Ok just wanted to update this, I have since corrected my script above to run on my raspberry pi, jessie release:

```
```
#! /bin/sh
# /etc/init.d/picam

### BEGIN INIT INFO
# Provides: picam
# Required-Start: $remote_fs $syslog $all
# Required-Stop: $remote_fs $syslog
# Default-Start: 2 3 4 5
# Default-Stop: 0 1 6
# Short-Description: picam
# Description: picam
### END INIT INFO

case "$1" in
    start)
        echo "Starting picam"
        sudo mkdir /run/shm/hooks
        sudo mkdir /run/shm/state
        sudo mkdir /run/shm/rec
        sudo /usr/src/picam/picam --tcpout tcp://127.0.0.1:8181
    ;;
    stop)
        echo "Stopping picam"
        kill picam
    ;;
    *)
        echo "Usage: /etc/init.d/picam {start|stop}"
        exit 1
    ;;
esac
exit 0
```

iizukanao, 2016-01-09:
@FWCJohn Change the line as follows, then check /var/log/picam.log for error messages.

    /usr/src/picam/picam --tcpout tcp://127.0.0.1:8181 >/var/log/picam.log 2>&1

Also, make sure that ffmpeg is listening on tcp://127.0.0.1:8181 before executing picam.

hvanbakel, 2017-01-08:
@iizukanao I've been trying to get this to work using your predefined setup. While that is certainly cleaner with the separate config file I could not get it to work with audio and HLS (hosted in nginx).
It all starts fine but without sound. Then when I log into the pi and run
sudo service picam restart
I get a HLS stream with audio. 

If I use the script from @FWCJohn above and just paste in the full command it works upon booting with audio.

Great work by the way. It really works like a charm!

walterav1984, 2018-06-01:
Modified @FWCJohn his init script to fix kill(serivce stop) and deamonize& (silent error and log) it but using it to run as a controllable service (not integrated with boot yet) which already let me stream 24 hours to [tvheadend](www.tvheadend.org) :+1: . Notice that when using .asoundrc for pi user it has to be copied to /root/ or a /etc/asound.con had to be made to replace the user .asoundrc, otherwise no sound! 

```
#! /bin/sh
# /etc/init.d/picam

### BEGIN INIT INFO
# Provides: picam
# Required-Start: $network
# Required-Stop: $network
# Default-Start: 2 3 4 5
# Default-Stop: 0 1 6
# Short-Description: picam
# Description: picam
### END INIT INFO

case "$1" in
    start)
        echo "Starting picam"
        sudo mkdir -p /run/shm/hooks
        sudo mkdir -p /run/shm/state
        sudo mkdir -p /run/shm/rec
        /home/pi/picam-1.4.6-binary-stretch/picam --alsadev "lp" -c 1 --time --timeformat "%Y/%m/%d birdbox-g %H:%M:%S" --timelayout top,left -f 25 -w 1440 -h 1080 -v 3000000 --hflip --vflip --wb off --wbred 1.1 --wbblue 1.9 --iso 500 --tcpout tcp://192.168.1.8:8181?listen >/var/log/picam.log 2>&1 &
    ;;
    stop)
        echo "Stopping picam"
        killall picam
        killall -9 picam
    ;;
    *)
        echo "Usage: /etc/init.d/picam {start|stop}"
        exit 1
    ;;
esac
exit 0
```
For tvheadend the following ffmpeg script needs to made and used with a IPTV configuration linking to the ffmpeg script "pipe:///home/hts/rpi0cam.sh".

Example /home/hts/rpi0cam.sh
```
#!/bin/bash
#start picam service before staring tvheadend stream/recording
/usr/bin/ffmpeg -loglevel quiet -f mpegts -i tcp://192.168.1.8:8181 -codec copy -f mpegts pipe:1
```
Just notice I reversed TCP listen direction since tvheadend is a bit random in initiating the stream and therefor hard the start the picam afterwards.