iizukanao/picam #32: Displaying multiple text overlays simultaneously

Linkaan, 2016-01-02:
Hi Nao,

I have switched over to a RPi 2 because the text overlay was unfortunately too CPU intensive for an RPI A+. This feature is very useful to me and I think it could be further improved by having the option to display multiple text overlays simultaneously. I'd like to be able to write a subtitle hook in this manner:
```
{
text=●REC
font_name=FreeSans
pt=40
layout_align=top,right
horizontal_margin=30
vertical_margin=30
color=000000
stroke_width=0
duration=0
}
{
text=OUTSIDE 25'C\\nCPU 30'C
font_name=FreeMono:style=Bold
pt=20
layout_align=top,left
text_align=left
horizontal_margin=30
vertical_margin=30
duration=0
}
``` 
Or even to have a directory `subtitle` to which you can write multiple subtitle hooks dynamically.
I understand this might be hard to implement and there are workarounds I could do to overcome this but if it is possible to implement I think it would be a good improvement!

Another improvement I'd love to see which could be very useful is to have an option, like `show_in_video=true|false`, to display the text overlay in the recorded video or only on the livestream. This I understand might be extremely difficult to implement and it is not really necessary. I think it is somehow possible to add overlays in ffmpeg dynamically anyway?

iizukanao, 2016-01-05:
Thanks for the suggestion. That would be a nice improvement! Overlaying text with ffmpeg should be a last resort.

paolohu, 2016-03-14:
i put my vote in for this one - i'd also like to see the option to show the overlay on the live preview (if possible)

bmatusiak, 2020-08-25:
i know this is old, but +1,  using hooks,   something to add for recommendation of api, 

current api hook example
```
echo 'text=Houston, we have a problem' > hooks/subtitle
```

my recommendation for multi overlays
```
echo 'text=Houston, we have liftoff' > hooks/subtitle_a
echo 'text=Houston, we have a problem' > hooks/subtitle_b
```