iizukanao/picam #33: o-o-o-o.... output on raspbian jessie lite

armadaazteca, 2016-01-15:
I was running the binary 1.4.1 version on raspbian whezzy and then I started a new installation based on jessie lite, after adding a couple of libraries I was able to start picam, but in the output I get o-o-o-o-o-o-o- all the time, why is that?

this is the command I'm running: 
**./picam --rtspout --alsadev hw:1,0 --volume 2**

and the output:
```
configuring devices
o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-capturing started
o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-o-
```

iizukanao, 2016-01-17:
I've just tested on Jessie lite, and picam worked fine. Do you have similar issues when you record video with `raspivid`, or record audio with `arecord`?

iizukanao, 2016-01-17:
Oh sorry, I've found the culprit. `o-` and `o+` means audio clipping (the volume is too high). Reduce the `--volume 2` value.

armadaazteca, 2016-01-17:
Awesome thanks I will give it a try, there might be something wrong with my sound card, I started getting errors like:

```
microphone error: wait for poll failed
trying to recover from error
microphone error: wait for poll failed
trying to recover from error
microphone error: wait for poll failed
trying to recover from error
microphone error: wait for poll failed
trying to recover from error
microphone error: wait for poll failed
trying to recover from error
```

Anyway, reducing the volume fixed the issue. Thanks!!

closed 2016-01-17 by armadaazteca