iizukanao/picam #34: Issue with Microphone after 15-30 mins of use

FWCJohn, 2016-01-17:
Ever since I have installed the picam, I am getting about 15 minutes of good audio and visual stream. The system then goes into a mass write over and over suddenly out of the blue, writing to system-journal this, 40 times a second:

Jan 17 05:57:05 raspberrypi picam[595]: microphone error: wait for poll failed
Jan 17 05:57:05 raspberrypi picam[595]: trying to recover from error

SDometimes it continues on like this forawhile and then suddenly starts again. Sometimes it never stops and I have to reboot. Usually it is writing this so much, it throws the load avg to 2 almost, locking up the machine.

FWCJohn, 2016-01-17:
FYI I have also tested it with another microphone, lasted a bit longer but around 20 minutes in, it just locks up writing that to log.

iizukanao, 2016-01-17:
Could you please run the following command and post the results? This tests the SD card write speed by writing a 500 MB file.

    $ dd if=/dev/zero of=~/test.tmp bs=500K count=1024

armadaazteca, 2016-01-17:
sorry to steal the thread, but since I'm also having the same issue, if it helps I will leave here the output of my system:

```
~/picam-1.4.1-binary $ dd if=/dev/zero of=~/test.tmp bs=500K count=1024
1024+0 records in
1024+0 records out
524288000 bytes (524 MB) copied, 27.9073 s, 18.8 MB/s
```

I haven't done the symlinks on the hsm, do you think this would help?

iizukanao, 2016-01-18:
@armadaazteca Hmm, 18.8 MB/s seems fine. Can you record audio with `arecord` command without any problems? Symlinks might make a difference when picam is recording to a file.

armadaazteca, 2016-01-19:
I think I found the culprit. The sound card for some reason is disconnecting over and over, so I guess the polling fails during this disconnects and floods the logs with the error. I'm checking to see if the disconnects are due to some kind of power saving setting or an issue with the driver/kernel combination. I have tested the same sound card in a couple of raspberry pi 2s and I see the same entries on the message logs, just as FYI, these are the ones, I found this because the same thing happened with the wifi card, and after upgrading the firmware the issue stopped on the wifi USB device, but persist on the sound card:

```
Jan 19 05:35:35 raspberrypi kernel: [190887.149395] usb 1-1.2: USB disconnect, device number 6
Jan 19 05:35:35 raspberrypi kernel: [190887.467519] usb 1-1.2: new full-speed USB device number 7 using dwc_otg
Jan 19 05:35:35 raspberrypi kernel: [190887.569918] usb 1-1.2: New USB device found, idVendor=0d8c, idProduct=000c
Jan 19 05:35:35 raspberrypi kernel: [190887.569946] usb 1-1.2: New USB device strings: Mfr=0, Product=1, SerialNumber=0
Jan 19 05:35:35 raspberrypi kernel: [190887.569963] usb 1-1.2: Product: C-Media USB Headphone Set
Jan 19 05:35:35 raspberrypi kernel: [190887.605545] input: C-Media USB Headphone Set   as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.3/0003:0D8C:000C.0003/input/input3
Jan 19 05:35:35 raspberrypi kernel: [190887.658145] hid-generic 0003:0D8C:000C.0003: input,hidraw0: USB HID v1.00 Device [C-Media USB Headphone Set  ] on usb-3f980000.usb-1.2/input3

Jan 19 05:35:55 raspberrypi kernel: [190907.120210] usb 1-1.2: USB disconnect, device number 7
Jan 19 05:35:55 raspberrypi kernel: [190907.437758] usb 1-1.2: new full-speed USB device number 8 using dwc_otg
Jan 19 05:35:55 raspberrypi kernel: [190907.540075] usb 1-1.2: New USB device found, idVendor=0d8c, idProduct=000c
Jan 19 05:35:55 raspberrypi kernel: [190907.540103] usb 1-1.2: New USB device strings: Mfr=0, Product=1, SerialNumber=0
Jan 19 05:35:55 raspberrypi kernel: [190907.540120] usb 1-1.2: Product: C-Media USB Headphone Set
Jan 19 05:35:55 raspberrypi kernel: [190907.576072] input: C-Media USB Headphone Set   as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.3/0003:0D8C:000C.0004/input/input4
Jan 19 05:35:55 raspberrypi kernel: [190907.628368] hid-generic 0003:0D8C:000C.0004: input,hidraw0: USB HID v1.00 Device [C-Media USB Headphone Set  ] on usb-3f980000.usb-1.2/input3

Jan 19 05:36:35 raspberrypi kernel: [190947.317415] usb 1-1.2: USB disconnect, device number 8
Jan 19 05:36:35 raspberrypi kernel: [190947.638209] usb 1-1.2: new full-speed USB device number 9 using dwc_otg
Jan 19 05:36:35 raspberrypi kernel: [190947.740690] usb 1-1.2: New USB device found, idVendor=0d8c, idProduct=000c
Jan 19 05:36:35 raspberrypi kernel: [190947.740719] usb 1-1.2: New USB device strings: Mfr=0, Product=1, SerialNumber=0
Jan 19 05:36:35 raspberrypi kernel: [190947.740736] usb 1-1.2: Product: C-Media USB Headphone Set
Jan 19 05:36:35 raspberrypi kernel: [190947.774538] input: C-Media USB Headphone Set   as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.3/0003:0D8C:000C.0005/input/input5
Jan 19 05:36:36 raspberrypi kernel: [190947.828963] hid-generic 0003:0D8C:000C.0005: input,hidraw0: USB HID v1.00 Device [C-Media USB Headphone Set  ] on usb-3f980000.usb-1.2/input3
```

Linkaan, 2016-01-19:
@armadaazteca maybe it helps if you try to use a USB hub with an external power supply.

armadaazteca, 2016-01-20:
I don't have one, but I do have one of those cheap USB tester monitors, and I don't see the pi consuming even close to 1A (I'm using a 2A source), so I don't think this is the problem.

FWCJohn, 2016-01-21:
@armadaazteca Thanks this is the exact issue I am having as well. Where did you find "firmware" upgrades for the pi?

armadaazteca, 2016-01-22:
you can use these 2 commands
```
# Check firmware version
/opt/vc/bin/vcgencmd version

# Upgrade firmware
sudo rpi-update
```

Basically the firmware is taken from a repository in github, either from the *master* or the *next* branch. you need to make sure the update tool is updated, so first you can run:

```
sudo apt-get update rpi-update
```

niklasdoerfler, 2016-07-08:
Hey, 
I've got a similar problem.
PiCam runs fine, sound is also working and synchronous to video.
But after some time (usually some hours) the microphone error appears and I have to restart the program, or sometime even the raspberry pi.

I let PiCam run in verbose mode, with the following output:

```
......................... 24.01 fps k=6466 a-v=26640 c-a=2576128 u=250 d=1 pts=606583920
......................... 24.07 fps k=6467 a-v=26970 c-a=2575616 u=250 d=1 pts=606677670
......................... 24.04 fps k=6468 a-v=27300 c-a=2575104 u=250 d=1 pts=606771420
......................... 24.08 fps k=6469 a-v=25710 c-a=2576448 u=250 d=1 pts=606865170
..................trying to recover from error
microphone error: buffer overrun
[microphone started]..trying to recover from error
microphone error: buffer overrun
[microphone started]..trying to recover from error
microphone error: buffer overrun
[microphone started]trying to recover from error
microphone error: buffer overrun
[microphone started]trying to recover from error
microphone error: buffer overrun
[microphone started]trying to recover from error
microphone error: buffer overrun
[microphone started].trying to recover from error
microphone error: buffer overrun
[microphone started]trying to recover from error

...

microphone error: buffer overrun
[microphone started]trying to recover from error
microphone error: buffer overrun
[microphone started]trying to recover from error
microphone error: buffer overrun
[microphone started]trying to recover from error
microphone error: buffer overrun
[microphone started]trying to recover from error
microphone error: buffer overrun
[microphone started]trying to recover from error
.microphone error: buffer overrun
[microphone started]Killed
```

I'm running PiCam on an Raspberry Pi 2+ with newest Raspbian and latest Raspberry Firmware. The sounddevice is also an "C-Media Electronics, Inc. Audio Adapter".

Does anybody know how to solve the problem?

Thanks in advance!
Niklas

Frodox, 2016-12-26:
+1. Also have `C-Media Electronics, Inc. Audio Adapter` and same problem.
@iizukanao i have 
`524288000 bytes (524 MB) copied, 57.725 s, 9.1 MB/s` - which speed is minimum required?

midworld, 2016-12-31:
I have same issue. I think it isn't power problem because my device has enough power.

iizukanao, 2017-01-01:
@niklasdoerfler There is a hint in your console output.
```
warn: fps > 20 might not work properly when width and height is large.
      Use lower --fps or use --vfr. If you still want to use this
      configuration, see if picam keeps up with 25.0 fps using --verbose.
```

iizukanao, 2017-01-01:
@Frodox It depends on the video bitrate, but generally that write speed is fine.

Miljar, 2017-04-04:
Still suffering from the `buffer overrun` issue:

<details>

    pi@phpwvl-cam:~ $ ./picam/picam --alsadev hw:0,0 --fps 25 --tcpout tcp://127.0.0.1:8181 --verbose
    video_width=1280
    video_height=720
    video_fps=25.000000
    fr_q16=1638400
    video_pts_step=3600
    video_gop_size=25
    video_rotation=0
    video_hflip=0
    video_vflip=0
    video_bitrate=2000000
    video_avc_profile=constrained_baseline
    video_avc_level=3.1
    video_qp_min=-1
    video_qp_max=-1
    video_qp_initial=-1
    video_slice_dquant=-1
    alsa_dev=hw:0,0
    audio_channels=1
    audio_sample_rate=48000
    audio_bitrate=40000
    audio_volume_multiply=1.000000
    is_hlsout_enabled=0
    is_hls_encryption_enabled=0
    hls_encryption_key_uri=stream.key
    hls_encryption_key=0x75b0a81de17487c88a47507a7e1fdf73
    hls_encryption_iv=0x000102030405060708090a0b0c0d0e0f
    hls_output_dir=/run/shm/video
    rtsp_enabled=0
    rtsp_video_control_path=/tmp/node_rtsp_rtmp_videoControl
    rtsp_audio_control_path=/tmp/node_rtsp_rtmp_audioControl
    rtsp_video_data_path=/tmp/node_rtsp_rtmp_videoData
    rtsp_audio_data_path=/tmp/node_rtsp_rtmp_audioData
    tcp_enabled=1
    tcp_output_dest=tcp://127.0.0.1:8181
    auto_exposure_enabled=0
    auto_exposure_threshold=5.000000
    is_vfr_enabled=0
    white_balance=auto
    exposure_control=auto
    awb_red_gain=0.000000
    awb_blue_gain=0.000000
    metering=average
    manual_exposure_compensation=0
    exposure_compensation=0.000000
    manual_exposure_aperture=0
    exposure_aperture=0
    manual_exposure_shutter_speed=0
    exposure_shutter_speed=0
    manual_exposure_sensitivity=0
    exposure_sensitivity=0
    roi_left=0.000000
    roi_top=0.000000
    roi_width=1.000000
    roi_height=1.000000
    min_fps=-1.000000
    max_fps=-1.000000
    is_timestamp_enabled=0
    timestamp_format=%a %b %d %l:%M:%S %p
    timestamp_layout=15
    timestamp_horizontal_margin=10
    timestamp_vertical_margin=10
    is_timestamp_abs_pos_enabled=0
    timestamp_pos_x=0
    timestamp_pos_y=0
    timestamp_text_align=1
[... 111 lines cut ...]
    shutdown_openmax: disable tunnel[0]
    shutdown_openmax: teardown tunnels
    shutdown_openmax: state transition to idle
    shutdown_openmax: state transition to loaded
    shutdown_openmax: ilclient_cleanup_components
    shutdown_openmax: OMX_Deinit
    shutdown_openmax: ilclient_destroy cam_client
    shutdown_openmax: ilclient_destroy ilclient
    shutdown_video
    teardown_audio_encode
    teardown_audio_capture_device
    hls_destroy
    pthread_mutex_destroy
    teardown_tcp_output
    teardown_socks
    free_encoded_packets
    stop_watching_hooks
    stop requested (signal=15)
    pthread_join hooks_thread
    free keyframe_pointers
    shutdown successful

</details>

I'm using a microphone connected to a USB sound card on a RPi Zero W. Is there any more useful information I can provide to get this solved?

roopannava, 2017-11-01:
I had the same issue :microphone error: buffer overrun
[microphone started]trying to recover from error

After I changed a new USB microphone everything is working fine.

sakshirawal14, 2018-03-13:
Hello everyone
I'm having the same issue. I tested the code yesterday since morning and somewhere in the evening, I started receiving the microphone error: buffer overrun

How can it be resolved? What causes the error?