iizukanao/picam #35: newbie question

dfischer123, 2016-02-14:
trying to get this to work with my limited programming skills. i get stuck at the "To start recording, create a file named hooks/start_record while picam command is running." ive created the symlinks however, when i start ./picam i dont get a prompt it just says capturing started and i can only exit it with ctrl c. please help.

iizukanao, 2016-02-15:
Hi, thanks for using picam!
That is normal. picam never returns while it is working. It says "capturing started" when everything is fine. Keep picam running in separate terminal, then open a new terminal and run the touch command.

dfischer123, 2016-02-15:
thanks! got it working, however, when i use echo -e command to try and record to a mounted nas folder, its output is 0 while the recording rests in picam/rec/archive. how do i permanently have picam record to /mnt/nas which is my mounted nas drive folder. your work and help is awesome!

iizukanao, 2016-02-15:
Make picam/rec/archive a symlink to /mnt/nas with the following commands.

    $ cd picam/rec
    $ mkdir old_archive
    $ mv archive/* old_archive
    $ rmdir archive
    $ ln -s /mnt/nas archive

Then try recording again.

dfischer123, 2016-02-15:
now a file is created in /mnt/nas with the timestamp.ts but it has 0 bytes no matter how long i run start_record for.any ideas?

iizukanao, 2016-02-15:
As long as picam generates files in picam/rec/archive/ without problems, it must be a problem in the NAS side. Re-examine your NAS configuration. What is the filesystem of NAS? What is the write speed?

iizukanao, 2016-02-15:
Alternatively, you can move around the recorded files with `mv` command after a recording is completed.

dfischer123, 2016-02-15:
was a permissions issue. all working now. really appreciate your help. many thanks!

closed 2016-02-15 by dfischer123