iizukanao/picam #38: Displaying frame number

timdegouss, 2016-03-08:
Hi Nao,

Is it possible to display frame number in the timestamp ? It seems there is no flag for frame in strftime format specification. Do you know a way to do that ? 

thank you very much !

timdegouss, 2016-03-08:
(The main goal behind this is to find a way to synchronise pi video with an external SMPTE signal)

iizukanao, 2016-03-08:
Yes, it's technically possible, but not implemented yet. Will just a specifier like `%f` enough for you?

timdegouss, 2016-03-08:
il will be perfect !
thank you so much :+1:

timdegouss, 2016-03-08:
And to be very complete is it possible to add the timestamp (with frame number) of the first frame into a text file? Maybe in file state/recorded_filename or somewhere else. 
This way it will be easy to embed timecode into video file with ffmmpeg flag -timecode

iizukanao, 2016-03-08:
OK, I'll do that when I have time.