iizukanao/picam #39: Arch Linux "Sample format" errors

tylerjl, 2016-03-25:
I'm trying to get picam to work on a raspberry pi running Arch Linux Arm.

The precompiled version doesn't work since the binary is linked against an older version of libpng:

```
$ picam-1.4.1-binary/picam --help
picam-1.4.1-binary/picam: error while loading shared libraries: libpng12.so.0: cannot open shared object file: No such file or directory
$ ls -l /usr/lib/libpng*
lrwxrwxrwx 1 root root     11 Jan 24 17:34 /usr/lib/libpng.so -> libpng16.so
lrwxrwxrwx 1 root root     19 Jan 24 17:34 /usr/lib/libpng16.so -> libpng16.so.16.21.0
lrwxrwxrwx 1 root root     19 Jan 24 17:34 /usr/lib/libpng16.so.16 -> libpng16.so.16.21.0
-rwxr-xr-x 1 root root 174108 Jan 24 17:34 /usr/lib/libpng16.so.16.21.0
```

So I built it from source. Unfortunately, there are still issues when trying to capture video:

```
$ ./picam --noaudio
configuring devices
Sample format s16 is not supported
```

(I don't have an audio device.) So I'm not really sure where the issue may lie - I can provide additional information if needed.

Thanks for the help!

iizukanao, 2016-03-25:
I would like to investigate the problem, but where did you download Arch Linux image? It seems Raspberry Pi's Downloads page no longer provides Arch Linux.

tylerjl, 2016-03-25:
This would have come from [Arch Linux Arm's download page](https://archlinuxarm.org/about/downloads), I'm on an armv6h Pi model.

iizukanao, 2016-03-27:
I've made some changes to the code. Could you please build the latest source and run picam again?

btw I've installed Arch Linux and built picam from source, and it worked without problems.

tylerjl, 2016-04-02:
My apologies, this was entirely user error on my end. For future reference, getting ffmpeg installed with libfdk fixed the problem. Arch users can install the `ffmpeg-libfdk_aac` package from the AUR (although it obviously takes some time to compile on the Pi), but after installing it, compiling and running picam works without issue. 

Thanks!

closed 2016-04-02 by tylerjl