iizukanao/picam #40: Subtitle - Not Transparent

paolohu, 2016-04-01:
Have you ever thought about adding a background color parameter to the text overlay?  I use the overlays to display the score of a game.  It would be nice to just overlay white text over a black background.

Linkaan, 2016-04-29:
This would indeed be a nice improvement.

iizukanao, 2016-04-29:
@paolohu Sorry for the delay. Thanks for your suggestion! I'm considering adding this feature in the next version.

paolohu, 2016-05-13:
another possibility here - or maybe a different enhancement - i could see overlaying an image onto the video - like a logo or other graphic like a scoreboard, etc.

i'm not sure about the older pi's but it seems like the pi3 should have enough processing power.  right now picam seems to be using right around 10% of the CPU on the pi3 when recording/streaming 1080p video.