iizukanao/picam #41: secondary low resolution stream?

kdish, 2016-04-15:
First of all, thanks for creating and sharing picam, it works really well.

I have used your picam-streamer pre-built image (https://github.com/iizukanao/picam-streamer)  to implement a motion detecting stecurity camera, using a PIR sensor connected to a GPIO port and a small Python script that will use the hooks file API to start recording whenever motion is detected by the sensor. It only took me minutes to get this to work the way I wanted, and I'm no real programmer, only a hobbyist. 
I use the rtsp live stream to access the camera from my Android units. This works fine while connected via WiFi to the same network as the camera, but I would like to be able to view the live stream remotely from my Andriod phone as well. The default video settings of 1280x720@30fps are a bit high for this to work smoothly. I am aware that I can lower the resolution and frame rate, but that will also impact the recordings..... 
I have dabbled a bit with transcoding and re-streaming the rstp stream on another Linux machine using VLC, ffmpeg, gstreamer etc., but I have a hard time finding a reliable solution.

I noticed that the "picamera" Python module  is able to provide two simultaneous x264 streams  at different resolutions (see. https://picamera.readthedocs.org/en/release-1.10/recipes2.html#recording-at-multiple-resolutions), and my question is if this is something you would like to consider as well? I.e. make picam capable of outputting an optional secondary low resolution stream, suitable for viewing over lower bandwidth connections.

iizukanao, 2016-04-17:
Hi, thanks for using picam!

Thanks for letting me know the details. I'm now considering to support this feature in future versions.

kdish, 2016-04-17:
Thanks! Just let me know if I can help in any way with testing or anything else...

kdish, 2016-04-17:
Sorry didn't mean to close the issue.

paolohu, 2016-05-12:
so would the multiple streams work for something like - i want to stream at 720p but then also record at 1080p?

kdish, 2016-05-13:
My thinking was to only provide a secondary lower resolution stream and have everything else work like now, i.e the primary stream is used for recording and the secondary stream is only available via the streaming intefaces (rtmp,rstp, HLS). That scenario would support your example. You could set up the primary stream at 1080p and connect to the secondary stream at 720p. My use case would be a much lower resolution secondary stream, possible to view over a mobile network connection.

kcunko, 2017-07-31:
It would be nice to have possibility to stream secondary stream to v4l2loopback video device