iizukanao/picam #42: Embed via JW Player

bloigge, 2016-04-16:
Hi,

First of all: great application (both picam and node-server). I have a problem viewing the livestream. I am using the JW Player and viewing files from the file directory works great.

the URL I use is: `rtmp://192.168.0.21/file/someFile.mp4`

But when I try using picam starting the application with:

`./picam --alsadev hw:1,0 --rtspout`

and using the following URL:

`rtmp://192.168.0.21/live/picam`

it errors with:

`[rtmp:client=9atPYDEo] error: stream not found: live/picam`

iizukanao, 2016-04-18:
You should see an output like this in the terminal of node server:

    2016-04-19 03:56:22.031 [rtsp] internal stream name has been set to: picam

If you are not seeing this message, "live/picam" endpoint is not available. Please run node server first, then start picam.

Alternatively, you can use [picam-streamer](https://github.com/iizukanao/picam-streamer) which is handy SD card image for live streaming.

bloigge, 2016-04-19:
[image]

Then I guess picam and node server are Ok.

Guess then something with my JW Player settings must be wrong - thats my code - any guess?:

`<script type="text/javascript">
    $(document).ready(function(){        
        jwplayer('mediaplayer').setup({
            'playlist': [{
              'sources': [{ 
        			file: 'rtmp://192.168.0.21/live/picam'
        		}]
            }],
            'width': 480,
            'height': 320,
            'primary': 'flash',
        });
    });
</script>`

When I press play pm2 logs the following:

[image]

timdegouss, 2016-04-19:
I think I have a similar issue. I recently update my scripts with the latest version of picam and node-server. Get the stream with the adress live/picam doesn't work anymore. 

With VLC, previously I was using this address: **rtsp://ip_adress:80/live/picam**. Now it doesn't work. I have to use this one instead: **rtsp://ip_adress:80/picam**.

iizukanao, 2016-04-22:
Sorry, that was my mistake. The issue has been fixed at ac8559e4ce92bf577ba30d093bc42b9e4bb63f12. Please pull the latest master source and build it.

closed 2016-04-26 by iizukanao