iizukanao/picam #43: Start and stop tcpout via hooks

ldamman, 2016-05-24:
Hello,

Thanks for the picam project!  I have been using some custom scripts calling ffmpeg to stream to ustream for a couple years now and I have never been able to get the audio to sync up with the video.  In picam, this just works!  Amazing.

Now for my request.  Would it be possible to add hooks to start_tcpout and stop_tcpout?  These would work similar to start_record and stop_record, except instead of controlling the recording to disk, they would control sending (or stopping) the stream to nginx.

iizukanao, 2016-06-22:
Thank you! This would be a nice improvement.

kcunko, 2017-07-29:
The same thing as for preview - it would be nice to have control via hooks to start and stop display preview.