iizukanao/picam #44: Picam + FFMpeg + tcpout + --vfr = stalling

CloakedAlien, 2016-06-17:
I know this could possible be a bit vague and might not even be a picam issue but an issue with ffmpeg, but please bare with me =)

Goal: really, really low-latency streaming from PI to Wowza

Solution: ffmpeg in listening mode + picam. Works like a charm!

Issues: can't go below 1024x768 resolution because ffmpeg stalls and picam throws errors, but only when using variable framerate. 1024x768 does give is quite good results however, even with audio. 

Whenever it fails picam spits out *"warning: Record buffer is starving. Recorded file may not start from keyframe. Try reducing the value of --gopsize."* 
More often than not it will not terminate with ctrl-c but needs to be killed.
We're not recording anything, simply sending it to ffmpeg.

Ffmpeg sometimes reports issues reading the video data but correctly reports the metadata. It shows the framerate being extremely low when failing. Sound seems to always go through!

**ffmpeg commandline:**
ffmpeg -i tcp://127.0.0.1:8181?listen -c:v copy -c:a aac -strict -2 -ar 44100 -ab 64000 -f flv rtmp://ourwowzaserver

**picam commandlines:**
*works:*
./picam --alsadev hw:1,0 -c 1 --volume 0.1 -r 44100 -a 64000 --fps 20 -v 1000000 -w 640 -h 480 --hflip --vflip --tcpout tcp://127.0.0.1:8181

*won't work:*
./picam --alsadev hw:1,0 -c 1 --volume 0.1 -r 44100 -a 64000 --vfr -v 1000000 -w 640 -h 480 --hflip --vflip --tcpout tcp://127.0.0.1:8181

*works:*
./picam --alsadev hw:1,0 -c 1 --volume 0.1 -r 44100 -a 64000 --vfr  --maxfps 30 --minfps 10 --shutter 20000 -v 1000000 -w 1024 -h 768 --hflip --vflip --tcpout tcp://127.0.0.1:8181

*won't work:*
./picam --alsadev hw:1,0 -c 1 --volume 0.1 -r 44100 -a 64000 --vfr  --maxfps 30 --minfps 10 --shutter 20000 -v 1000000 -w 800 -h 600 --hflip --vflip --tcpout tcp://127.0.0.1:8181

The --vfr and --shutter settings seems to be crucial for me to achieve really low latency (mostly from trial 'n' error). 

FFMpeg is freshly compiled on the RPI itself:
ffmpeg version N-80344-g6fd4145 Copyright (c) 2000-2016 the FFmpeg developers
  built with gcc 4.9.2 (Raspbian 4.9.2-10)
  configuration: --arch=armel --target-os=linux --enable-gpl --enable-libx264 --enable-nonfree
  libavutil      55. 24.100 / 55. 24.100
  libavcodec     57. 46.100 / 57. 46.100
  libavformat    57. 38.100 / 57. 38.100
  libavdevice    57.  0.101 / 57.  0.101
  libavfilter     6. 46.101 /  6. 46.101
  libswscale      4.  1.100 /  4.  1.100
  libswresample   2.  1.100 /  2.  1.100
  libpostproc    54.  0.100 / 54.  0.100

Picam is the 1.4.1 binary, but I think I've had the same issue with 1.3.0

If you have any insights or ideas they would be greatly appreciated! =)

CloakedAlien, 2016-06-17:
Okay, here's another fine one.

*Works:*
./picam --alsadev hw:1,0 -c 1 -r 48000 -a 64000 --volume 0.2 --shutter 20000 -w 1024 -h 768 --hflip --vflip --tcpout tcp://127.0.0.1:8181

*Doesn't work:*
./picam --alsadev hw:1,0 -c 1 -r 48000 -a 64000 --volume 0.2 --shutter 20000 -w 1024 -h 768 --hflip --vflip **--wb off --metering backlit** --tcpout tcp://127.0.0.1:8181

iizukanao, 2016-06-22:
@CloakedAlien Thank you for reporting the issue in detail!

I think the problem has been fixed at the latest commit in master branch. If you can build picam from source, could you please check if it works for you?

CloakedAlien, 2016-06-23:
I'm a bit of a rookie but I've given it a shot =)

First of all I never did get the cross-compilation of ffmpeg to work properly. however I already have a working version of ffmpeg installed on my raspberry pi (the Pi3 actually builds ffmpeg in about an hour of you compile using make -j4). Does the actual compilation of picam require any of the files resulting from the ffmpeg cross-compilation or should I be fine as I already have a working version?

Anywho, I figured I'd try to just compile picam and after picking off the dependency errors I arrived at this:
`$ make

cc -c stream.c -o stream.o -DSTANDALONE -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -D_LINUX -fPIC -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -Wall -g -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -ftree-vectorize -pipe -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -Wno-psabi -I/opt/vc/include/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -I/opt/vc/src/hello_pi/libs/ilclient `freetype-config --cflags` `pkg-config --cflags harfbuzz fontconfig` -I/usr/include/fontconfig -g -Wno-deprecated-declarations -O3 -pipe -march=armv6zk -mfpu=vfp -mfloat-abi=hard

stream.c: In function âsetup_av_frameâ:

stream.c:686:49: error: âAVFrameâ has no member named âchannelsâ
   log_debug("av_frame->channels: %d\n", av_frame->channels);
                                                 ^
Makefile:16: recipe for target 'stream.o' failed`

iizukanao, 2016-07-24:
@CloakedAlien Terribly sorry for the late reply. I've just published version 1.4.2 binary, and it includes this fix. Please give it a try.

tomph, 2018-03-04:
Did this fix your issue @CloakedAlien ?

I'm seeing issues where if I use --vfr --ex  auto (could but unrelated), but when I quickly turn a light on and off (on camera), the framerate drops and the streams freezes up (in desktop and in an app I'm testing).

Would really like to know how to maybe get around it (I don't need vfr) as picam is otherwise amazing.

Thanks for all of your work

tomph, 2018-03-04:
./picam --alsadev hw:1,0 -c 1 --volume 0.1 -r 44100 -a 64000 --vfr --maxfps 30 --minfps 10 --shutter 20000 -v 1000000 -w 1024 -h 768 --hflip --vflip --tcpout tcp://127.0.0.1:8181

Thanks to the above, this works for me...  Had to re-jig my resolution as I saw in the docs/log that <= 1.45 ratio is best for minfps/maxfps