iizukanao/picam #49: Recording "raspistill" like preview jpeg image while recording TS?

kcunko, 2016-09-20:
Trying to combine picam and linphone on rpi, and only way to do it is to connect linphone video to static picture source instead of camera (camera can not be shared btw picam and linphone), so I'm missing "raspistill" feature on picam - i.e. when hook "start_record" is executed to enable in addition constant generation of always the same image that will hold latest camera preview (i.e. "preview.jpg").

Other approach would be to introduce hooks "start_preview" and "stop_preview" to enable streaming to jpeg file (that should be possible to work concurrently with start_record and stop_record hooks).

Not-optimal approach would be to introduce hook "get_preview_image".

Is there possibility to introduce this feature?

paolohu, 2016-09-20:
Not sure how your are using picam - but if you are passing it off to FFMPEG (streaming for example) then you could have a second output defined on FFMPEG that would write out an image every so often

kcunko, 2016-09-20:
Thnx for suggestion, tried that, it works but it requires too much processing (Raspberry) resulting with slow refresh rate on linphone...  I'm using picam for cam&audio record, and in parallel I want to run linphone (SIP video and two-way audio connection) with the same camera picture/audio picam is recording in bg. Since picam is already receiving stream from camera and in one processing point has already jpeg image inside buffer, optimal way regarding CPU would be to introduce option/switch in picam to write that jpeg down while recording video.
I guess I could also use mediastreamer that would use picam video stream and stream it in linphone but that would be even more CPU unfrendly than ffmpeg option.

kcunko, 2016-10-31:
Any chance to implement this enhancement (as picam command line option)? Or perhaps to give a suggestion how It could be implemented. Will it be possible to open port 72 for Camera output (along with 71) and than tunnel it to  image_encode component input port, and then write out image_encode buffer into file? I'm not sure if camera ports 71 and 72 can work in parallel though...

closed 2017-02-17 by kcunko