iizukanao/picam #50: Connection to tcp://127.0.0.01:8181 failed: Connection refused

slegouffe, 2016-09-28:
Hi ! 

First thank you for picam, it is really good and simple, but, but, but I've an issue when I execute this command : ./picam --tcpout tcp://127.0.0.01:8181 an error occurred : 
[tcp @ 0x277f270] Connection to tcp://127.0.0.01:8181 failed: Connection refused
avio_open for tcp://127.0.0.01:8181 failed: Connection refused

Did I miss something ?

(PS : I try the same command with "sudo" but still have the same error).

slegouffe, 2016-09-29:
Answer to myself : 
Ok before ./picam --tcpout we have to call ffmpeg command  :
ffmpeg -i tcp://127.0.0.1:8181?listen -c:v copy -c:a aac -strict -2 -ar 44100 -ab 64000 -f flv $RTPM_SERVER

iizukanao, 2016-09-29:
@slegouffe You're right!

closed 2016-09-29 by iizukanao

balkrishna, 2022-08-02:
Hi !

Thanks once again for picam. However, I still get this error, even after following the steps mentioned above.

Here is the error dump I get:

[1:28:24.258127599] [13321]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3384-44d59841

[1:28:24.290931740] [13345]  WARN RPI raspberrypi.cpp:1202 Mismatch between Unicam and CamHelper for embedded data usage!

[1:28:24.291517362] [13345]  INFO RPI raspberrypi.cpp:1313 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media1

[1:28:24.292120909] [13321]  INFO Camera camera.cpp:1028 configuring streams: (0) 1920x1080-YUV420

[1:28:24.292404850] [13345]  INFO RPI raspberrypi.cpp:738 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1920x1080-SBGGR10_1X10 - Selected unicam format: 1920x1080-pBAA

[1:28:24.401111539] [13354]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Sharpness = 1.000000

[1:28:24.401191130] [13354]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Saturation = 1.000000

[1:28:24.401222148] [13354]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Contrast = 1.000000

[1:28:24.401250778] [13354]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ColourGains = [ 0.000000, 0.000000 ]

[1:28:24.401279962] [13354]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AwbMode = 0

[1:28:24.401305221] [13354]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeExposureMode = 0

[1:28:24.401329758] [13354]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureTime = 0

[1:28:24.401353554] [13354]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AwbEnable = true

[1:28:24.401377091] [13354]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]

[1:28:24.401405164] [13354]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Brightness = 0.000000

[1:28:24.401431312] [13354]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ScalerCrop = (680x692)/1920x1080

[1:28:24.401458386] [13354]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0

[1:28:24.401481515] [13354]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 1

[1:28:24.401877121] [13345]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0

[1:28:24.401908268] [13345]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0

[1:28:24.401930694] [13345]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0

[1:28:24.401950823] [13345]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0

[1:28:24.401971212] [13345]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0

[1:28:24.401990490] [13345]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0

[tcp @ 0x331f4b0] Connection to tcp://[127.0.0.1:8181](http://127.0.0.1:8181/) failed: Connection refused

avio_open for tcp://[127.0.0.1:8181](http://127.0.0.1:8181/) failed: Connection refused

iizukanao, 2022-08-04:
@balkrishna Hi, can you confirm if video recording is possible with the [libcamera-vid](https://www.raspberrypi.com/documentation/accessories/camera.html#libcamera-vid) command?

balkrishna, 2022-08-05:
@iizukanao yes, I can record video with libcamera-vid command

iizukanao, 2022-08-05:
@balkrishna OK. Could you please provide the steps to reproduce the problem with ffmpeg and picam?

balkrishna, 2022-08-05:
@iizukanao 

I have added the following to my nginx.conf:

`rtmp {
    server {
        listen 1935;
        chunk_size 4000;
        application webcam {
            #Enable livestreaming
            live on;
                
            #Disable recording
            record off;

            #allow

            exec /usr/bin/ffmpeg -i tcp://127.0.0.1:8181?listen -c:v copy -ar 44100 -ab 40000 -f flv rtmp://localhost:1935/webcam/mystream;
        }
    }
}
`

Further, I run , as mentioned above,  `ffmpeg -i tcp://127.0.0.1:8181?listen -c:v copy -ar 44100 -ab 40000 -f flv rtmp://localhost:1935/webcam/mystream` 

which exits with a prompt:

`Input #0, mpegts, from 'tcp://127.0.0.1:8181?listen':
  Duration: N/A, start: 0.021333, bitrate: N/A
  Program 1 
    Metadata:
      service_name    : Service01
      service_provider: FFmpeg
    Stream #0:0[0x100]: Video: h264 (Baseline) ([27][0][0][0] / 0x001B), yuv420p(progressive), 1920x1080, 30 fps, 30 tbr, 90k tbn, 60 tbc
    Stream #0:1[0x101]: Audio: aac (LC) ([15][0][0][0] / 0x000F), 48000 Hz, mono, fltp, 39 kb/s
[tcp @ 0xcdcde0] Connection to tcp://localhost:1935 failed: Connection refused
[rtmp @ 0xdeb1f0] Cannot open connection tcp://localhost:1935
rtmp://localhost:1935/webcam/mystream: Connection refused
`

iizukanao, 2022-08-06:
@balkrishna The error message
```
[tcp @ 0xcdcde0] Connection to tcp://localhost:1935 failed: Connection refused
[rtmp @ 0xdeb1f0] Cannot open connection tcp://localhost:1935 rtmp://localhost:1935/webcam/mystream: Connection refused
```
means that ffmpeg cannot connect to the RTMP server (rtmp://localhost:1935/webcam/mystream), and it is not a problem of ffmpeg or picam. Please check if your RTMP server settings are correct.

I will lock this issue because the original issue is not relevant to your case. Please open a new issue.