iizukanao/picam #53: error creating state dir (./state): File exists

apastel, 2016-10-27:
I get this error when attempting to start picam for the second time.

I installed picam exactly according to the readme, by creating the make_dirs.sh script, which sets up all the directories and the links. But when running picam subsequent times, it fails to start because `state` already exists.

So I delete `state`. Then I try to start picam and get an error that `hooks` already exists.

So basically, every time I want to start up picam again, I have to delete all the directory links and re-run make_dirs.sh.

Is anyone else having this problem? Is there something I'm missing in the way I have this set up?

Running Raspian Jessie (lite) on a Raspberry Pi 3.

iizukanao, 2016-10-27:
@apastel I think you've messed up something in your environment. picam won't stop because of "state already exists". It will only create prerequisite directories if they don't exist.
Try deleting **rec**, **hooks**, **state**, and **archive** directories from picam, then run picam without make_dirs.sh and see if it works.

apastel, 2016-10-31:
Thanks. I tried that and it seems to be working now.

closed 2016-11-07 by iizukanao