iizukanao/picam #55: Playing on Android Devices

rodcoelhao, 2016-11-08:
Hi,

first, congrats. The picam and node-rtmp projects are great.

In my current setup I run node-rtmp and picam on the rpi and play the stream on a Mac with Strobe Media Player. I have tried other players (including VLC) but the latency is quite large(>5s)

I have tried to play on an android device (One PlusX) but with no luck. I installed the dolphin browser, adobe flash and swf player. Whenever I open the html file with dolphin I can see the player with a black rectangle with the blue progress bar and all controls but I cannot see or hear anything.

I can open the ramp stream directly with VLC on my android device, but with large latency.

Do you have any experience on playing it with Strobe Media Player on android? Any further pointer?

Thanks,

iizukanao, 2016-12-05:
@rodcoelhao With [this app](https://play.google.com/store/apps/details?id=air.tv.lotusfusion.rtmp.player.one) I get about two seconds of latency. I don't think Flash on Android yields better results.

viveksanagari, 2019-02-07:
Hi @rodcoelhao 
I was trying to see the stream on Android as well. However, I was not able to view the RTMP stream on mobile with strobe media player initially. This was because my android didn't support adobe flash plugin on the browser, a strobe player uses flash. 

The strobe media player can fallback for devices without flash, and it can be done using HLS streaming. We can find this at the end of this `https://github.com/iizukanao/picam/blob/master/test/strobe_media_playback.html` 
file as,
```
<div id="StrobeMediaPlayback">
  <!-- fallback for devices without Flash is HLS streaming -->
  <video src="http://PI_HOST/hls/index.m3u8" width="480" height="360" autoplay controls></video>
</div>
```

I was using Nginx with the picam. So to be able to see the stream on android I have configured the nginx.conf file for HLS. Which looked like this,
```...
http {
...
   server {
               ...
               location /hls/ {
		        root /run/shm;
		        index index.html index.htm index.m3u8;
	      }
              ...
    }
}
rtmp {
        server {
                listen 1936;
                chunk_size 4096;
		
		application webcam {
			live on;
			
			hls on;
			hls_path /run/shm/hls;
		}
        }
}
```

After this nginx configuration and replacing the IP address at
```
<video src="http://PI_HOST/hls/index.m3u8" width="480" height="360" autoplay controls></video>
```
to the IP of the Raspberry Pi, I was able to view the stream in android & iOS.

viveksanagari, 2019-02-07:
# Update

Hi again, there was a case for me once,  where even after the above configuration I was not able to see the stream on android. 

As the **hls_path** is mentioned as **/run/shm/hls** in the nginx.conf file. I tried to see what was in the folder **/run/shm/hls** after starting the video capture.
using, `ls /run/shm/hls`, there were files being saved with name as **mystream.ts** and so. (As **mystream** was the stream key for my stream)
So the stream was present but not showing up on the webpage.

The problem was the name specified at the 'video src'. As the files were saving as **mystream.ts** and the `src` is specified with `index.m3u8` in `<video src="http://PI_HOST/hls/index.m3u8" width="480" height="360" autoplay controls></video>` there was a mismatch in the names.

So specifing the name with which they were being saved at **/run/shm/hls** in the 'src' solved my problem. The correct 'src' for me looked like:
`<video src="http://PI_HOST/hls/mystream.m3u8" width="480" height="360" autoplay controls></video>`

Hope this helps someone. :)

joelihn, 2020-09-10:
> # Update
> Hi again, there was a case for me once, where even after the above configuration I was not able to see the stream on android.
>
> As the **hls_path** is mentioned as **/run/shm/hls** in the [...]

Why "mystream",where to set this string?

pi@raspberrypi:~ $ ls /run/shm/hls
10.ts  13.ts  16.ts  19.ts  539.ts  542.ts  545.ts  548.ts  9.ts
11.ts  14.ts  17.ts  20.ts  540.ts  543.ts  546.ts  549.ts  enc.key
12.ts  15.ts  18.ts  21.ts  541.ts  544.ts  547.ts  550.ts  index.m3u8

On my rpi, the filename start with number.