iizukanao/picam #56: Timelapse support?

Geefer, 2016-11-14:
Would it be possible to add the option to output the .ts file when recording as a timelapse movie (writing only one frame in every 'n' where 'n' is configurable) so that the size of movie recorded can be reduced? 

I know this can be done as a post process with ffmpeg but that still requires the full movie to be recorded to flash and then converted, plus the conversion process takes a while on the Raspberry Pi. 

In timelapse mode audio would probably have to be removed I am guessing.

iizukanao, 2016-12-05:
@Geefer You can create a timelapse (low fps) video with `--fps` option (e.g. `--fps 0.5`). To silence audio, use `--noaudio` option together.

Geefer, 2016-12-05:
Brilliant! For some reason I never thought of doing it like that. 
Thanks!

Geefer, 2016-12-05:
Actually having tried it and investigated more I find with --fps 0.5 the streamed image is either totally black or frozen and sometimes it causes picam to lock up (I am using the picam-streamer setup)

What I really would ideally like is to be able to stream at full fps but only record every n frames so that the streaming is smooth and full speed but the recorded video is timelapse to save disk space.

iizukanao, 2016-12-05:
@Geefer Hmm, maybe it's possible with the help of ffmpeg. Use `--tcpout` option to stream the video from picam to ffmpeg, then create a timelapse video with ffmpeg in some way.