iizukanao/picam #58: How to output picam stream to a remote node-rtsp-rtmp-server server?

helxsz, 2016-11-30:
when installing picam and node-rtsp-rtmp-server server on the same PI platform, it works out by `./picam --rtspout ` that the picam automatically links the stream to the  node-rtsp-rtmp-server server on the pi. What if I have the node-rtsp-rtmp-server on a remote cloud server with certain IP address? what command I should use?

iizukanao, 2016-12-05:
You can do it with the help of ffmpeg. First, leave the following command running in Pi, where REMOTE_IP is the IP address of node-rtsp-rtmp-server.
```
$ ffmpeg -i tcp://127.0.0.1:8181?listen -c:v copy -c:a aac -ar 44100 -ab 40000 -f flv rtmp://REMOTE_IP/live/picam
```
Then, in another terminal window, run picam with `--tcpout tcp://127.0.0.1:8181` option. The streaming will begin after 10 seconds or so.

helxsz, 2016-12-05:
I followed the ffmpeg command above,   the node-rtsp-rtmp-server side (on another computer in the local area network) received an error:

```
root@tegra-ubuntu:/home/ubuntu/node-rtsp-rtmp-server# node server.js
2016-12-05 17:27:40.678 attachRecordedDir: dir=file app=file
2016-12-05 17:27:40.700 [rtmp] server started on port 1935
2016-12-05 17:27:40.701 [rtsp/http/rtmpt] server started on port 80
2016-12-05 17:28:08.831 [rtmp:client=aaPKdz69] connected
[rtmp: 1 sessions]
 aaPKdz69: addr=192.168.1.198 port=40512
2016-12-05 17:28:09.013 [rtmp:receive] unknown (not implemented) AMF command: _checkbw
2016-12-05 17:28:09.081 [rtmp] publish: stream=live/picam publishingType=live

/home/ubuntu/node-rtsp-rtmp-server/server.js:47
    throw err;
          ^
TypeError: Object 'B���n�x�5 has no method 'compare'
    at AVStream.updateSPS (/home/ubuntu/node-rtsp-rtmp-server/avstreams.js:206:49)
    at StreamServer.onReceiveVideoNALUnits (/home/ubuntu/node-rtsp-rtmp-server/stream_server.js:388:18)
    at StreamServer.onReceiveVideoPacket (/home/ubuntu/node-rtsp-rtmp-server/stream_server.js:413:12)
    at /home/ubuntu/node-rtsp-rtmp-server/stream_server.js:63:28
    at RTMPServer.emit (/home/ubuntu/node-rtsp-rtmp-server/rtmp.js:2666:20)
    at /home/ubuntu/node-rtsp-rtmp-server/rtmp.js:2599:31
    at RTMPSession.emit (/home/ubuntu/node-rtsp-rtmp-server/rtmp.js:1091:18)
    at /home/ubuntu/node-rtsp-rtmp-server/rtmp.js:2433:27
    at RTMPSession.handleData (/home/ubuntu/node-rtsp-rtmp-server/rtmp.js:2559:16)
    at Socket.<anonymous> (/home/ubuntu/node-rtsp-rtmp-server/rtmp.js:2620:34)
```

iizukanao, 2016-12-06:
@helxsz I think your Node.js version is too old. Download the latest version from https://nodejs.org/en/download/.

helxsz, 2016-12-06:
Thanks, It is the problem of old node.js, it works now.

closed 2016-12-06 by helxsz