iizukanao/picam #59: Little Problem with the rtmp

EscoR85, 2016-12-11:
hi,
first of all thank you for your work - it just works fine.
but:
i just can't get the part" Using picam in combination with nginx-rtmp-module" to work.

i installed ffmepg und the nginx-rtmp-module.
like you wrote in your guide, i edited the nginx.conf (/home/pi/nginx-1.2.4/config/nginx.conf)
`#user  nobody;
worker_processes  1;

error_log  logs/error.log debug;

events {
    worker_connections  1024;
}

http {
    include       mime.types;
    default_type  application/octet-stream;

    sendfile        on;
    keepalive_timeout  65;

    server {
        listen       8080;
        server_name  localhost;

        # sample handlers
        #location /on_play {
        #    if ($arg_pageUrl ~* localhost) {
        #        return 201;
        #    }
        #    return 202;
        #}
        #location /on_publish {
        #    return 201;
        #}

        #location /vod {
        #    alias /var/myvideos;
        #}

        # rtmp stat
        location /stat {
            rtmp_stat all;
            rtmp_stat_stylesheet stat.xsl;
        }
        location /stat.xsl {
            # you can move stat.xsl to a different location
            root /usr/build/nginx-rtmp-module;
        }

        # rtmp control
        location /control {
            rtmp_control all;
        }

        error_page   500 502 503 504  /50x.html;
        location = /50x.html {
            root   html;
        }
    }
}

rtmp {
    server {
        listen 1935;
        chunk_size 4000;
        application webcam {
            live on;

            exec_static /usr/local/bin/ffmpeg -i tcp://127.0.0.1:8181?listen
                                        -c:v copy -ar 44100 -ab 40000
                                        -f flv rtmp://localhost:1935/webcam/mystream;
        }
    }
}`

After that i start the server with `sudo /etc/init.d/nginx start`
than i start the ffmpeg.
`pi@raspberrypi:~ $ sudo ffmpeg -i tcp://127.0.0.1:8181?listen -c:v copy -c:a aac -strict -2 -ar 44100 -ab 64000 -f flv $RTPM_SERVER
ffmpeg version 2.7.2 Copyright (c) 2000-2015 the FFmpeg developers
  built with gcc 4.9.2 (Raspbian 4.9.2-10)
  configuration: --enable-gpl --enable-postproc --enable-swscale --enable-avfilter --enable-libmp3lame --enable-libvorbis --enable-libtheora --enable-libx264 --enable-libspeex --enable-pthreads --enable-libopenjpeg --enable-libfaac --enable-nonfree
  libavutil      54. 27.100 / 54. 27.100
  libavcodec     56. 41.100 / 56. 41.100
  libavformat    56. 36.100 / 56. 36.100
  libavdevice    56.  4.100 / 56.  4.100
  libavfilter     5. 16.101 /  5. 16.101
  libswscale      3.  1.101 /  3.  1.101
  libswresample   1.  2.100 /  1.  2.100
  libpostproc    53.  3.100 / 53.  3.100
Trailing options were found on the commandline.
Input #0, mpegts, from 'tcp://127.0.0.1:8181?listen':
  Duration: N/A, start: 0.021333, bitrate: N/A
  Program 1 
    Metadata:
      service_name    : Service01
      service_provider: FFmpeg
    Stream #0:0[0x100]: Video: h264 (Constrained Baseline) ([27][0][0][0] / 0x001B), yuv420p, 1280x720, 90k tbr, 90k tbn, 180k tbc
    Stream #0:1[0x101]: Audio: aac (LC) ([15][0][0][0] / 0x000F), 48000 Hz, mono, fltp, 11 kb/s
At least one output file must be specified
`

iizukanao, 2016-12-12:
@EscoR85 When you use nginx-rtmp-module, you don't have to manually run ffmpeg. Just restart nginx then run picam.

EscoR85, 2016-12-12:
hi, thank you for your fast replay.
i think i found the issue, my ffserver.conf doesn't work.
after `ffserver`
i get a mass of errors. is it possible to post your ffserver.conf? 
i think i would take a massive invest of time to figure out what is wrong.

EscoR85, 2016-12-17:
just did work out for me :(.
so i installed the out-of-the-box-sd version. works fine... thank you for this programm.
but there is just a tiny thing. :)
if i what to add some options to the picam, like --hflip, --autoex  etc,
where should i put it in?
in the  /home/pi/picam-streamer/lib/server.coffee ?

iizukanao, 2016-12-17:
@EscoR85 Sorry, I was busy. For the former question, I don't use ffserver, but I use ffmpeg. So, I don't have ffserver.conf.
To modify command line options of picam-streamer, please refer to [this comment](https://github.com/iizukanao/picam-streamer/issues/3#issuecomment-244641119).

EscoR85, 2016-12-17:
thx for your help works exellent.