iizukanao/picam #61: Problem with live streaming

jahanzeb07, 2016-12-31:
Hi, first thank you for such a nice program. Currently I'm using it to record videos with sound and its working fantastic. I am newbie in raspberry pi and couldn't figure out how to stream the video using your program. I followed the instructions in ReamMe file but may be I am not getting some thing. 
To stream video first I did    `$ ./picam -o /run/shm/hls` and then I put the IP address of my Pi into another PC in the same network like `http://RPIip/run/shm/hls/index.m3u8`. I did same in my PI browser but it didn't work and I think I am doing right. You explained it quit well but I need step by step instructions.
Thank you for the help.

iizukanao, 2016-12-31:
@jahanzeb07 To enable HLS (HTTP Live Streaming), you also have to serve */run/shm/hls/* directory with an HTTP server such as nginx or Apache. Then, you can view the stream at http://RPIip/index.m3u8.

iizukanao, 2017-01-01:
@jahanzeb07 See [this page](https://www.raspberrypi.org/documentation/remote-access/web-server/nginx.md) for instructions. You don't need to do the section "ADDITIONAL - INSTALL PHP".

jahanzeb07, 2017-01-02:
Hi, thank you for your assistance. I followed the instructions from the page you referred and successfully installed the nginx server and checked it after starting and it was working. Then, I followed the nginx documentation and I added following block into nginx.conf file:
`server {
     location /hls/ {
     index index.html index.m3u8;
     root /run/shm/hls;`

but still it is not working. I don't know what I am doing wrong. I did google for the problem but couldn't find solution.
Thank you for your time and support.

iizukanao, 2017-01-02:
@jahanzeb07 Replace `root /run/shm/hls` with `root /run/shm` like this:
```
        location /hls/ {
                root /run/shm;
                index index.m3u8 index.html;
        }
```
Then, you can access the HLS stream at http://localhost/index.m3u8 or just http://localhost/.

jahanzeb07, 2017-01-03:
Hi, thank you again. I tried with your code but still no success. I got the same error, 404 not found.
I am posting here everything so you may figure out what I am doing wrong.
In  /etc/nginx/nginx.conf file within http section I added:
`server {
                location /hls/ {
                root /run/shm;
                index index.html index.m3u8 index.htm;
                }
        }
`
I restart the nginx server by

`sudo service nginx restart` and `sudo nginx -s reload`

then,

In picam directory I did,

`./picam -o /run/shm/hls`

output of the command is

`pi@raspberrypi:~/picam $ ./picam -o /run/shm/hls/
configuring devices
error: cannot open audio capture device 'hw:0,0': No such file or directory
hint: specify correct ALSA device with '--alsadev <dev>'
warning: audio capturing is disabled
capturing started
`

and 

`ls -l /run/shm/hls/
total 9908
-rw-r--r-- 1 pi pi 279368 Jan  3 11:53 184.ts
-rw-r--r-- 1 pi pi 281060 Jan  3 11:53 185.ts
-rw-r--r-- 1 pi pi 284444 Jan  3 11:53 186.ts
-rw-r--r-- 1 pi pi 285384 Jan  3 11:53 187.ts
-rw-r--r-- 1 pi pi 280496 Jan  3 11:53 188.ts
-rw-r--r-- 1 pi pi 281060 Jan  3 11:53 189.ts
-rw-r--r-- 1 pi pi 285008 Jan  3 11:53 190.ts
-rw-r--r-- 1 pi pi 283316 Jan  3 11:53 191.ts
-rw-r--r-- 1 pi pi 282940 Jan  3 11:53 192.ts
-rw-r--r-- 1 pi pi 281812 Jan  3 11:53 193.ts
-rw-r--r-- 1 pi pi 283692 Jan  3 11:53 194.ts
-rw-r--r-- 1 pi pi  70876 Jan  3 11:53 195.ts
-rw-r--r-- 1 pi pi 284444 Jan  3 11:44 203.ts
-rw-r--r-- 1 pi pi 288392 Jan  3 11:44 204.ts
-rw-r--r-- 1 pi pi 279368 Jan  3 11:44 205.ts
-rw-r--r-- 1 pi pi 283504 Jan  3 11:44 206.ts
-rw-r--r-- 1 pi pi 279368 Jan  3 11:44 207.ts
-rw-r--r-- 1 pi pi 285196 Jan  3 11:44 208.ts
-rw-r--r-- 1 pi pi 278428 Jan  3 11:44 209.ts
-rw-r--r-- 1 pi pi 287452 Jan  3 11:44 210.ts
-rw-r--r-- 1 pi pi 286888 Jan  3 11:44 211.ts
-rw-r--r-- 1 pi pi 280872 Jan  3 11:44 212.ts
-rw-r--r-- 1 pi pi 282376 Jan  3 11:44 213.ts
-rw-r--r-- 1 pi pi  94940 Jan  3 11:44 214.ts
-rw-r--r-- 1 pi pi 280496 Jan  3 12:07 75.ts
-rw-r--r-- 1 pi pi 280872 Jan  3 12:07 76.ts
-rw-r--r-- 1 pi pi 288392 Jan  3 12:07 77.ts
-rw-r--r-- 1 pi pi 282564 Jan  3 12:07 78.ts
-rw-r--r-- 1 pi pi 282000 Jan  3 12:07 79.ts
-rw-r--r-- 1 pi pi 283316 Jan  3 12:07 80.ts
-rw-r--r-- 1 pi pi 281624 Jan  3 12:07 81.ts
-rw-r--r-- 1 pi pi 284068 Jan  3 12:07 82.ts
-rw-r--r-- 1 pi pi 283316 Jan  3 12:07 83.ts
-rw-r--r-- 1 pi pi 280120 Jan  3 12:07 84.ts
-rw-r--r-- 1 pi pi 282000 Jan  3 12:07 85.ts
-rw-r--r-- 1 pi pi 287076 Jan  3 12:07 86.ts
-rw-r--r-- 1 pi pi 271660 Jan  3 12:07 87.ts
-rw-r--r-- 1 pi pi    142 Jan  3 12:07 index.m3u8
`
I tried to access the live streaming in web browser by [http://localhost](url) but it gives me default page. I also tried with [http://localhost/hls](url) , [http://localhost/hls/index](url) , [http://localhost/index.m3u8](url) and also [http://localhost/run/shm/hls](url) but all these give me error. Still I don't know what is wrong.
Thank you for your patience and guidance.

iizukanao, 2017-01-03:
@jahanzeb07 Can you show me the full contents of your /etc/nginx/nginx.conf?

jahanzeb07, 2017-01-03:
Hi, thank you very much for the update in ReadMe file. I just saw your update and I got now what I was doing wrong. I was adding location in `/etc/ngnix/ngnix.conf `only but not in `/etc/nginx/sites-available/default` . I added the code in default file also and now it is working. I can watch the stream from my Pi using VLC Player. Is it possible to watch the live stream from web browser? 

Thank you again for your time.

closed 2017-01-03 by jahanzeb07

iizukanao, 2017-01-03:
@jahanzeb07 Great! You can use https://github.com/dailymotion/hls.js to play HLS in a browser.

feacluster, 2019-09-13:
Using apache I just had to do this: 

cd /var/www/html/
mkdir hls
ln -s /run/shm/hls hls