iizukanao/picam #62: Running as a child process

vGoldsmith, 2017-01-16:
Hello,

I recently started playing with picam to learn c, I'm having trouble passing arguments to picam via execv().  This is what I tried:

`
static char *argv[]={"--alsadev", "hw:1,0","-w 1920","-h 1080","-f 30",NULL};
execv("/home/pi/picam/picam",argv);
`

I get the error:

> error: cannot open audio capture device 'hw:0,0': No such file or directory
hint: specify correct ALSA device with '--alsadev <dev>'
warning: audio capturing is disabled

What I'm trying to do is set up a key press event to start picam as a child process; furthermore, start recording audio and video.  Also, hopefully, another key-press event to end it and execute another program.  At the moment I'm hung up on the above.

I appreciate you time and help, it's a great repo

nalajcie, 2017-01-16:
Hi,
your problem is not with picam, but within Your C code, so it would be better not to post a new issue regarding picam project :).

Nevertheless, let me help You a little. You've created invalid argv table. Read the man docs (http://man7.org/linux/man-pages/man2/execve.2.html), there is an example there.
Things You're missing:
- _By convention, the first of these strings (i.e., argv[0]) should contain the filename associated with the file being executed_ [from man], so `argv[0`] should be `"picam"`
- each and every param is a separate string

Your array should be something about:
`static char *argv[]={"picam", "--alsadev", "hw:1,0", "-w", "1920", "-h", "1080", "-f", "30", NULL};`

Additional hint: use `strace` to check if the attributes to execv* functions are the same as issuing them from the command line, for example:
`$ strace prog -s xx --somearg asdfasdf
execve("/usr/bin/prog", ["prog", "-s", "xx", "--somearg", "asdfasdf"], [/* 72 vars */]) = 0
[..]`

I suggest You close the issue and ask C-related questions on appropriate websites (stackoverflow, forums, etc.) as this is not related to picam project.

vGoldsmith, 2017-01-16:
Thank you!  Sorry about that.

closed 2017-01-16 by vGoldsmith