iizukanao/picam #63: Microphone buffer overflow / Record buffer is starving

leobeosab, 2017-01-17:
Hi all, I keep getting a microphone buffer overflow or a record buffer starving when recording and random intervals. Originally it was just the microphone buffer overflow but I changed the audio buffer size in PiCam and recompiled to have record buffer is starving show up instead. Any way to fix or at least have the camera continue to record when microphone buffer overflow happens so I don't have 5 seconds missing from my video? Thanks!

Here is my command:

`/home/pi/software/picam --verbose --time --timeformat "%b %e, %Y %l:%M:%S %p %Z" --width 1280 --height 720 --videobitrate 3000000 --fps 30.0 --alsadev hw:1,0 --samplerate 44100 --recordbuf 1`

macsimc, 2017-01-18:
Hi, i am having the same issue, but i am not sure if it is related to the microphone recording since it failed on me today while testing with --noaudio after 30-60s.
It is strange that sometimes with the exact same settings it works for several hours.
When it happens the process needs to be killed, maybe it would be better if it would crash so it could be automatically restarted.

............... 15.07 fps k=213 a-v=-6927 c-a=4427 u=0 d=0 pts=19176000
............... 15.10 fps k=214 a-v=-9168 c-a=6666 u=0 d=0 pts=19266000
............... 15.05 fps k=215 a-v=-7230 c-a=4598 u=0 d=0 pts=19356000
............... 15.01 fps k=216 a-v=-9471 c-a=6839 u=0 d=0 pts=19446000
............... 15.01 fps k=217 a-v=-7533 c-a=4853 u=0 d=0 pts=19536000
............... 15.00 fps k=218 a-v=-9774 c-a=7106 u=0 d=0 pts=19626000
...............E(0x0).warning: Record buffer is starving. Recorded file may not start from keyframe. Try reducing the value of --gopsize.
warning: Record buffer is starving. Recorded file may not start from keyframe. Try reducing the value of --gopsize.
warning: Record buffer is starving. Recorded file may not start from keyframe. Try reducing the value of --gopsize.
^Cstop requested (signal=2)
shutdown sequence start
waiting for the camera to finish

picam  --verbose --time --recordbuf 1 --avcprofile baseline --avclevel 3.1 --metering matrix -c 2 -r 44100 -a 128000 -v 4000000 -f 15 -g 5  --width 1640 --height 1232 --alsadev hw:1,0 --tcpout tcp://127.0.0.1:8181

leobeosab, 2017-01-18:
Hm that would make sense, let me know if you have a solution for recording at > 2MBs. Thanks!

closed 2017-01-18 by leobeosab