iizukanao/picam #65: new stream name for node rtmp server seems to break playback on omxplayer and vlc

gepatto, 2017-01-28:
In stream.c the stream name has been changed to 'live/picam'
For me that breaks playback from node-rtsp-rtmp-server to omxplayer or vlcplayer,
as the the complete url now becomes rtmp://host/live/live/picam which omxplayer and vlcplayer don't seem to understand. (Strobe Media Playback still works)

Is there a reason why this was changed?

iizukanao, 2017-01-29:
@gepatto That change has been made long ago in order to fix interoperability with node-rtsp-rtmp-server. Are you using the latest version of node-rtsp-rtmp-server?

gepatto, 2017-01-29:
I'll check out the latest version of node-rtsp-rtmp-server and see if that fixes it.

gepatto, 2017-01-29:
Yes that fixed it. I was using node-rtsp-rtmp-server from the picam-streamer image..
The current git pull works!
Thank you for the support and the wonderful software!

closed 2017-01-29 by gepatto