iizukanao/picam #66: Alsa - Invalid Argument

paolohu, 2017-01-30:
I've been using picam quite awhile with a USB microphone.  I've been playing around with the audioinjector sound card (http://www.audioinjector.net/#!/rpi-hat).

The card seems to work find with arecord and aplay - and with ffmpeg using raspivid | ffmpeg.

When I try and start picam using it I get the following error:

Set video_encode state to executing
opening ALSA device for capture: hw:0,0
cannot use mono audio; trying stereo
final audio_channels: 2
sample_rate: 48000
nb_samples: 1024
sample_fmt: 1
audio_codec_ctx->channel_layout: 3
av_frame->channel_layout: 3
audio_codec_ctx->channels: 2
av_frame->channels: 0
allocated 4096 bytes for audio samples
audio_pts_step_base: 1920
error: cannot set access type (Invalid argument)

If I go back to the usb mic then it's fine - so I'm assuming it has something to to with the card - anything I can try to troubleshoot further?

paolohu, 2017-01-30:
more information - if i set the alsadev to: plughw:CARD=audioinjectorpi,DEV=0

it will startup (ie, not crash) but there's no actual audio and the video freezes.  For a comparison here's the log of the startup with the usb microphone and one with the audio card...

usb mic:

allocated 2048 bytes for audio samples
audio_pts_step_base: 1920
actual sample rate=48000 dir=0
microphone: buffer size: 204800 frames (channels=1 buffer_size=2048 multiply=100)
microphone: setting period size to 1024
actual_period_size=1024 dir=0
audio capture device:
Plug PCM: Hardware PCM card 1 'USB Audio Device' device 0 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : MMAP_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 1
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 204800
  period_size  : 1024
  period_time  : 21333
  tstamp_mode  : NONE
  period_step  : 1
  avail_min    : 1024
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 204800
  silence_threshold: 0
  silence_size : 0
  boundary     : 1677721600
  appl_ptr     : 0
  hw_ptr       : 0
start capturing video
waiting for the first video buffer
start capturing audio
[microphone started]capturing started

audio injector card:

opening ALSA device for capture: plughw:CARD=audioinjectorpi,DEV=0
final audio_channels: 1
sample_rate: 48000
nb_samples: 1024
sample_fmt: 1
audio_codec_ctx->channel_layout: 4
av_frame->channel_layout: 4
audio_codec_ctx->channels: 1
av_frame->channels: 0
allocated 2048 bytes for audio samples
audio_pts_step_base: 1920
actual sample rate=48000 dir=0
failed to set buffer size for microphone: buffer_size=2048 multiply=100
trying smaller buffer size for microphone: buffer_size=2048 multiply=50
microphone: buffer size: 102400 frames (channels=1 buffer_size=2048 multiply=50)
microphone: setting period size to 1024
actual_period_size=1024 dir=0
audio capture device:
Plug PCM: Route conversion PCM (sformat=S16_LE)
  Transformation table:
    0 <- 0*0.5 + 1*0.5
Its setup is:
  stream       : CAPTURE
  access       : MMAP_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 1
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 102400
  period_size  : 1024
  period_time  : 21333
  tstamp_mode  : NONE
  period_step  : 1
  avail_min    : 1024
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 102400
  silence_threshold: 0
  silence_size : 0
  boundary     : 1677721600
Slave: Mmap emulation PCM
Its setup is:
  stream       : CAPTURE
  access       : MMAP_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 102400
  period_size  : 1024
  period_time  : 21333
  tstamp_mode  : NONE
  period_step  : 1
  avail_min    : 1024
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 102400
  silence_threshold: 0
  silence_size : 0
  boundary     : 1677721600
Slave: Hardware PCM card 0 'audioinjector-pi-soundcard' device 0 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 102400
  period_size  : 1024
  period_time  : 21333
  tstamp_mode  : NONE
  period_step  : 1
  avail_min    : 1024
  period_event : 0
  start_threshold  : 1677721600
  stop_threshold   : 102400
  silence_threshold: 0
  silence_size : 0
  boundary     : 1677721600
  appl_ptr     : 0
  hw_ptr       : 0
start capturing video
waiting for the first video buffer
start capturing audio
[microphone started]not first audionot first

iizukanao, 2017-02-12:
@paolohu Looks like Audio Injector does not support mmap access. Can you set `mmap_emulation` to true in your ALSA config file?

paolohu, 2017-02-16:
i'm working through this one - right now i'm just getting:

Unknown field mmap_emulation

when I true and set that value in the config file.  I'll try a few more things and let you know.

paolohu, 2017-02-16:
So I couldn't figure out a way to make mmap_emulation work - but I did find a way to make the Audio Injector card support mmap.

In case it helps anyone in the future that uses that card (http://www.audioinjector.net) the fix is pretty simple:

Edit the /boot/config.txt file on the Raspberry Pi and include:

dtoverlay=i2s-mmap

once I did that everything started working properly with the picam app.

czarina, 2017-03-24:
I have this same issue. I'm using an i2s MEMS mic. I get:

'error: cannot set access type (Invalid argument)'

When I do ` sudo ./picam --alsadev mic_mono`

I have already added dtoverlay=i2s-mmap to /boot/config.txt but this did not help.

czarina, 2017-03-24:
Here is my verbose output:

$ sudo ./picam --alsadev mic_mono --verbose
video_width=1280
video_height=720
video_fps=30.000000
fr_q16=1966080
video_pts_step=3000
video_gop_size=30
video_rotation=0
video_hflip=0
video_vflip=0
video_bitrate=2000000
video_avc_profile=constrained_baseline
video_avc_level=3.1
video_qp_min=-1
video_qp_max=-1
video_qp_initial=-1
video_slice_dquant=-1
alsa_dev=mic_mono
audio_channels=1
audio_sample_rate=48000
audio_bitrate=40000
audio_volume_multiply=1.000000
is_hlsout_enabled=0
is_hls_encryption_enabled=0
hls_encryption_key_uri=stream.key
hls_encryption_key=0x75b0a81de17487c88a47507a7e1fdf73
hls_encryption_iv=0x000102030405060708090a0b0c0d0e0f
hls_output_dir=/run/shm/video
rtsp_enabled=0
rtsp_video_control_path=/tmp/node_rtsp_rtmp_videoControl
rtsp_audio_control_path=/tmp/node_rtsp_rtmp_audioControl
rtsp_video_data_path=/tmp/node_rtsp_rtmp_videoData
rtsp_audio_data_path=/tmp/node_rtsp_rtmp_audioData
tcp_enabled=0
tcp_output_dest=
auto_exposure_enabled=0
auto_exposure_threshold=5.000000
is_vfr_enabled=0
white_balance=auto
exposure_control=auto
awb_red_gain=0.000000
awb_blue_gain=0.000000
metering=average
manual_exposure_compensation=0
exposure_compensation=0.000000
manual_exposure_aperture=0
exposure_aperture=0
manual_exposure_shutter_speed=0
exposure_shutter_speed=0
manual_exposure_sensitivity=0
exposure_sensitivity=0
roi_left=0.000000
roi_top=0.000000
roi_width=1.000000
roi_height=1.000000
min_fps=-1.000000
max_fps=-1.000000
is_timestamp_enabled=0
timestamp_format=%a %b %d %l:%M:%S %p
timestamp_layout=15
timestamp_horizontal_margin=10
timestamp_vertical_margin=10
is_timestamp_abs_pos_enabled=0
timestamp_pos_x=0
timestamp_pos_y=0
timestamp_text_align=1
timestamp_font_name=FreeMono:style=Bold
timestamp_font_file=
timestamp_font_face_index=(null)
timestamp_font_points=14.000000
timestamp_font_dpi=96
timestamp_color=ffffff
timestamp_stroke_color=000000
timestamp_stroke_width=1
timestamp_letter_spacing=0
is_preview_enabled=0
is_previewrect_enabled=0
preview_x=0
preview_y=0
preview_width=0
preview_height=0
preview_opacity=255
blank_background_color=0x0
is_audio_preview_enabled=0
audio_preview_dev=plughw:0,0
record_buffer_keyframes=5
state_dir=state
hooks_dir=hooks
configuring devices
exposure control: auto
setting exposure:
  eMetering: 0
  xEVCompensation: 0
  nApertureFNumber: 0
  bAutoAperture: 0
  nShutterSpeedMsec: 0
  bAutoShutterSpeed: 1
  nSensitivity: 0
  bAutoSensitivity: 1
Set video_encode state to idle
Enable port buffers for camera 71
Enable port buffers for video_encode 200
Enable port buffers for video_encode 201
Set camera state to executing
Set video_encode state to executing
opening ALSA device for capture: mic_mono
final audio_channels: 1
sample_rate: 48000
nb_samples: 1024
sample_fmt: 1
audio_codec_ctx->channel_layout: 4
av_frame->channel_layout: 4
audio_codec_ctx->channels: 1
av_frame->channels: 0
allocated 2048 bytes for audio samples
audio_pts_step_base: 1920
error: cannot set access type (Invalid argument)