iizukanao/picam #67: ROI?

paolohu, 2017-02-01:
Is there anything in picam similar to the -roi (region of interest) functionality that exists in the raspivid app?

I am using a PiCapture HD1 board (emulates the camera) - my issue is certain resolutions (1080i for example) the full frame isn't video - so to work around it they recommend using -roi to pull the video.  It works fine for raspivid and it would be great if I could do something similar in picam.

thanks,

-paolo

iizukanao, 2017-02-12:
@paolohu Implemented `--roi` option in version 1.4.6. Please give it a try!

paolohu, 2017-02-13:
awesome - i'll give it a try today and let you know!

paolohu, 2017-02-16:
yep this works great!  appreciate the quick turnaround!

closed 2017-02-16 by iizukanao