iizukanao/picam #69: Can't set 720p60?

kamranayub, 2017-02-19:
I was reading the Pi Camera module V2 can support 720p with 60fps. If I try to set `-f 60.0` with Picam, it doesn't work... it actually seems to lock up the camera until a reboot.

```
pi@raspberrypi:~ $ picam/picam -f 60
configuring devices
error: failed to set camera to idle state
Perhaps another program is using camera, otherwise you need to reboot this pi
```

fcaraballo, 2020-10-09:
were you able to solve this? got the same issue, if I get this to work I will post it here

iizukanao, 2020-10-12:
Running picam at 1280x720@60fps leaves the camera in a stuck state, but 60fps works with lower resolutions like 400x300. Maybe we need to use different internal parameters for larger resolutions.