iizukanao/picam #71: Error setting alsa device name?

czarina, 2017-03-24:
I have the following ALSA device:

`pi@raspberrypi:~/picam $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: memsmic [mems-mic], device 0: bcm2835-i2s-ics43432-hifi ics43432-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0`

I try picam:

`pi@raspberrypi:~/picam $ sudo ./picam --alsadev memsmic
configuring devices
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM memsmic
error: cannot open audio capture device 'memsmic': No such file or directory
hint: specify correct ALSA device with '--alsadev <dev>'
warning: audio capturing is disabled
capturing started
disk_usage=23% start rec to rec/archive/2017-03-24_01-36-33.ts
stop rec`

The resulting capture does not have any audio. Why is my memsmic not being recognized?

czarina, 2017-04-07:
@iizukanao Hello, would you please let me know if you have a chance to review this issue? I would really appreciate it.

dariondol, 2017-05-03:
try `--alsadev  hw:1,1`

simspace-dr, 2018-11-14:
@czarina did you ever successfully resolve this issue?

simspace-dev, 2018-11-14:
@czarina did you ever successfully resolve this issue?