iizukanao/picam #74: Record and stream simultaneously

masoudr, 2017-05-13:
Hi,
First, thank you for the awesome project, I tested a lot of related solutions and this one is one of the best.
My questions maybe already were asked but I didn't find a straight forward answer yet.
I have 2 questions:
1. How can I record locally and stream with HLS simultaneously? what is the best method for that?
2. I've followed your instructions about making encryption for the HLS, I run the command with no problem and created the key as you mentioned, but I can only access the stream with HTTP, not HTTPS.
Thanks.

iizukanao, 2017-05-19:
@masoudr 
1) Run a web server (e.g. nginx or Apache) and make HLS output directory public.
2) HLS encryption has nothing to do with HTTPS. If you want to use HTTPS, you need to enable it with a web server.

masoudr, 2017-05-19:
Thanks for the answer.
1. Can you explain a little more specific or give me a reference about that.
2. OK, Can I add an authentication login? Can I use the general method for Nginx web server? I didn't find any solution on your topic about that.
Recently I got into another problem. I'm using pi zero 1.3 with a wifi dongle and a tiny USB mic.
when I stream the picture quality is okay, but the sound is noisy and makes crackles, the same combination on my Pi 3 is working flawlessly but with the built in wifi. I think the problem is with wifi dongle, but I changed it with another and the problem still was there. Can you help me about that? Does anything special about the bandwidth of the stream?

iizukanao, 2017-05-19:
1. Follow these instructions (you don't need php) https://www.raspberrypi.org/documentation/remote-access/web-server/apache.md
Then, run picam with `-o /var/www/html/hls` option. At this point, the HLS stream is available at http://YOUR_PI_IP/hls/index.m3u8

2. For authentication, Basic Auth is very easy to setup. Try googling it.

As for audio noise, WiFi dongle might cause such noise. Put microphone away from WiFi dongle. Also, please make sure that you are using a power supply that provides enough power.

masoudr, 2017-05-20:
Thank you.
I tried to take away the mic, and it was fixed then I tried to use my 4g USB dongle which has built in wifi too, but the noise getting crazy and got worse.
here is the log window: (I see a lot of "-o-o-o" in the terminal)
`...-o-o-o-o-o-o-o-o-o-trying to recover from error
microphone error: buffer overrun`
Is there any solution to fix it?

iizukanao, 2017-05-21:
Please lower the microphone volume with `alsamixer` command and see if "o-o-o-" disappears.