iizukanao/picam #79: HLS stream delay about 3s compare with RTSP

quancao, 2017-06-24:
Hi,
Thank you for great project.
I just use picam and compare the delay when streaming over HLS and RTSP.
I follow your guideline and enable both HLS and RTSP using this command
./picam --alsadev plughw:1,0 -o /run/shm/hls --samplerate 16000

In my PC, I use VLC to view both HLS and RTSP and I found that HLS stream always delay about 3s compare with RTSP stream.
Do you have any suggestion to lower this delay ?

Thanks,
Quan

iizukanao, 2017-06-24:
@quancao Unfortunately no, you can't reduce the delay as long as you use HLS. As I wrote in README:
> In HTTP Live Streaming (HLS), the latency will never go below 3-4 seconds. This limitation stems from the design of HLS.

quancao, 2017-06-24:
Thanks. I have another question. Do you have any recommendation for using picam with pulseaudio ?

iizukanao, 2017-06-24:
I have never used picam with PulseAudio. What would you like to do with it?

quancao, 2017-06-24:
I want to use picam as a doorbell camera which I can stream the video to my remote and talk to it (Same as IP camera). I think I could use PulseAudio which reuse its Echo Cancellation feature

iizukanao, 2017-06-25:
Sorry, I don't know much about PulseAudio, and picam only works with ALSA at this time. However, it seems you can [configure ALSA to use PulseAudio](https://wiki.archlinux.org/index.php/PulseAudio#ALSA).

natriumlamp, 2019-07-04:
Dear quancao
Have you been successful implementing Echo Cancellation in Picam?
I am currently thinking that the only way is put speex or opus in Picam source code and recompile it but I wonder if there's way using PulseAudio.

quancao, 2019-07-04:
Sorry. I just did some experiments with webrtc and it is quite good.

natriumlamp, 2019-07-05:
> Sorry. I just did some experiments with webrtc and it is quite good.
> […](#)

I see.
Thank you.

quancao, 2019-07-05:
I tried speex but seems performance not good. I use usb mic and usb speaker for testing.