iizukanao/picam #80: Buffer overrun - streaming stops after a few hours

TiborUdvari, 2017-07-03:
Hello, 

I'm trying to get a livestream working continuously with a raspberry pi 3B and `picam` and the `nginx-rtmp-module` for my [infinitefidgetspinner.com](infinitefidgetspinner.com) project. 

The streaming was working fine for about 18 hours before it stopped.
This is what I got in the console :

```
pi@raspberrypi:~/picam $ ./picam --alsadev hw:1,0 --tcpout tcp://127.0.0.1:818
configuring devices
capturing started
trying to recover from error
microphone error: buffer overrun
```

I'm not sure if the livestream crashed because of the microphone error or after. Is there any log file that I could check for more info? Or have a way to see when it fails to restart the process?

TiborUdvari, 2018-05-10:
No, I didn't find a solution in the meantime

sakshirawal14, 2018-05-15:
Okay..Thanks nonetheless!