iizukanao/picam #83: nothing happens after typing ./picam --alsadev hw:1,0

carlhung, 2017-08-17:
after typing ./picam --alsadev hw:1,0
i read the guide that i can type hooks/start_record to start.
and type hooks/stop_record to stop.
It was at the first time. after that, it won't work again. nothing shows no the screen and no file was created.

iizukanao, 2017-08-18:
@carlhung Have you tried rebooting your pi?
Also, what messages are printed when you run picam with `--verbose` option?

carlhung, 2017-08-19:
turned out , I removed and reinstalled raspbrian again. But I just bumped into other program as I downloaded the newest image for the pi. Which is not Jessie anymore, the new raspbian is based on Debian 9 whose codename is Stretch. When I ran picam, it showed the program can't find libGLESv2.so

iizukanao, 2017-08-22:
@carlhung Thank you for reporting the issue! I've just fixed it. Please download the updated file picam-1.4.6-binary-stretch.tar.xz from [v1.4.6 Binary Release](https://github.com/iizukanao/picam/releases/tag/v1.4.6).

carlhung, 2017-08-24:
I still can't record anything.
pi@pizero:~/picam $ ./picam --alsadev hw:1,0 --verbose
video_width=1280
video_height=720
video_fps=30.000000 
fr_q16=1966080
video_pts_step=3000 
video_gop_size=30   
video_rotation=0
video_hflip=0
video_vflip=0
video_bitrate=2000000
video_avc_profile=constrained_baseline
video_avc_level=3.1 
video_qp_min=-1
video_qp_max=-1
video_qp_initial=-1 
video_slice_dquant=-1
alsa_dev=hw:1,0
audio_channels=1
audio_sample_rate=48000
audio_bitrate=40000 
audio_volume_multiply=1.000000
is_hlsout_enabled=0 
is_hls_encryption_enabled=0
hls_encryption_key_uri=stream.key
hls_encryption_key=0x75b0a81de17487c88a47507a7e1fdf73
hls_encryption_iv=0x000102030405060708090a0b0c0d0e0f
hls_output_dir=/run/shm/video
rtsp_enabled=0
rtsp_video_control_path=/tmp/node_rtsp_rtmp_videoControl
rtsp_audio_control_path=/tmp/node_rtsp_rtmp_audioControl
rtsp_video_data_path=/tmp/node_rtsp_rtmp_videoData
rtsp_audio_data_path=/tmp/node_rtsp_rtmp_audioData
tcp_enabled=0
tcp_output_dest=
auto_exposure_enabled=0
auto_exposure_threshold=5.000000
is_vfr_enabled=0
white_balance=auto  
exposure_control=auto
awb_red_gain=0.000000
awb_blue_gain=0.000000
metering=average
manual_exposure_compensation=0
exposure_compensation=0.000000
manual_exposure_aperture=0
exposure_aperture=0 
manual_exposure_shutter_speed=0
exposure_shutter_speed=0
manual_exposure_sensitivity=0
exposure_sensitivity=0
roi_left=0.000000   
roi_top=0.000000
roi_width=1.000000  
roi_height=1.000000 
min_fps=-1.000000   
max_fps=-1.000000   
is_timestamp_enabled=0
timestamp_format=%a %b %d %l:%M:%S %p
timestamp_layout=15 
timestamp_horizontal_margin=10
timestamp_vertical_margin=10
is_timestamp_abs_pos_enabled=0
timestamp_pos_x=0   
timestamp_pos_y=0   
timestamp_text_align=1
timestamp_font_name=FreeMono:style=Bold
timestamp_font_file=
timestamp_font_face_index=(null)
timestamp_font_points=14.000000
timestamp_font_dpi=96
timestamp_color=ffffff
timestamp_stroke_color=000000
timestamp_stroke_width=1
timestamp_letter_spacing=0
is_preview_enabled=0
is_previewrect_enabled=0
preview_x=0
preview_y=0
preview_width=0
preview_height=0
preview_opacity=255 
blank_background_color=0x0
is_audio_preview_enabled=0
audio_preview_dev=plughw:0,0
record_buffer_keyframes=5
state_dir=state
hooks_dir=hooks
configuring devices 
exposure control: auto
setting exposure:   
  eMetering: 0
  xEVCompensation: 0
  nApertureFNumber: 0
  bAutoAperture: 0  
  nShutterSpeedMsec: 0
  bAutoShutterSpeed: 1
  nSensitivity: 0   
  bAutoSensitivity: 1
Set video_encode state to idle
Enable port buffers for camera 71
Enable port buffers for video_encode 200
Enable port buffers for video_encode 201
Set camera state to executing
Set video_encode state to executing
opening ALSA device for capture: hw:1,0
final audio_channels: 1
sample_rate: 48000  
nb_samples: 1024
sample_fmt: 1
audio_codec_ctx->channel_layout: 4
av_frame->channel_layout: 4
audio_codec_ctx->channels: 1
av_frame->channels: 0
allocated 2048 bytes for audio samples
audio_pts_step_base: 1920
actual sample rate=48000 dir=0
microphone: buffer size: 204800 frames (channels=1 buffer_size=2048 multiply=100)
microphone: setting period size to 1024
actual_period_size=1024 dir=0
audio capture device:
Hardware PCM card 1 'USB PnP Audio Device' device 0 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : MMAP_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 1  
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16 
  buffer_size  : 204800
  period_size  : 1024
  period_time  : 21333
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1  
  avail_min    : 1024
  period_event : 0  
  start_threshold  : 1
  stop_threshold   : 204800
  silence_threshold: 0
  silence_size : 0  
  boundary     : 1677721600
  appl_ptr     : 0  
  hw_ptr       : 0  
start capturing video
waiting for the first video buffer
start capturing audio
[microphone started]capturing started
.[NAL7]............................... 31.00 fps k=1 a-v=-840 c-a=1708 u=0 d=0 pts=93000
...................^[[B........... 29.99 fps k=2 a-v=-600 c-a=1554 u=0 d=0 pts=183000
.............................. 30.05 fps k=3 a-v=-360 c-a=1211 u=0 d=0 pts=273000

when I press ctrl+c:
.....................^Cstop requested (signal=2)
shutdown sequence start
waiting for the camera to finish
Reverting pBuffer to its original value
stop capturing video
shutdown_openmax: ilclient_flush_tunnels
shutdown_openmax: disable port buffer for camera 71
shutdown_openmax: disable port buffer for video_encode 200
shutdown_openmax: disable port buffer for video_encode 201
shutdown_openmax: disable tunnel[0]
shutdown_openmax: teardown tunnels
shutdown_openmax: state transition to idle
shutdown_openmax: state transition to loaded
shutdown_openmax: ilclient_cleanup_components
shutdown_openmax: OMX_Deinit
shutdown_openmax: ilclient_destroy cam_client
shutdown_openmax: ilclient_destroy ilclient
shutdown_video
teardown_audio_encode
teardown_audio_capture_device
hls_destroy
pthread_mutex_destroy
teardown_socks
free_encoded_packets
stop_watching_hooks 
pthread_join hooks_thread
stop requested (signal=15)
free keyframe_pointers
shutdown successful

carlhung, 2017-08-24:
$ ./picam --alsadev hw:1,0
configuring devices
capturing started
touch hooks/start_record // 10 secs later, I typed this line.
touch hooks/stop_record // 1 min later, T typed this line.

nothing showed on the screen. 
and I get into ~/picam/rec/archive, no file could be found.

iizukanao, 2017-08-25:
@carlhung You need to enter `touch hooks/start_record` and `touch hooks/stop_record` in another shell. Open new tab in your terminal while picam is running, `cd` to picam directory, and run those commands.

carlhung, 2017-08-25:
it works now. thanks.

carlhung, 2017-08-25:
is it possible to define output filename of the ts file?
as i want to wire it up and make a cam corder that provide a start button and stop button. once i press stop, I want my program to find that file and calling ffmpeg to convert immediately. sorry, i am new in programming. I don't know if i can do like that.

iizukanao, 2017-08-26:
@carlhung You cannot directly control the output filenames, but you can get the filename of latest ts file with `ls -1t rec/archive/*.ts | head -n1` and pass it to ffmpeg command. For example:
```sh
#!/bin/sh
latestfile=$(ls -1t rec/archive/*.ts | head -n1)
ffmpeg -i "$latestfile" <arguments...>
```

carlhung, 2017-08-26:
Thanks. I was thinking the same. Find the last file, convert it and remove the file immediately

closed 2017-08-28 by carlhung

sakshirawal14, 2018-03-10:
How can I run the following commands as a bash script. 
$ ./picam --alsadev hw:1,0 
touch hooks/start_record 
touch hooks/stop_record 
It stalls after running the first command and prints
configuring devices
capturing started

iizukanao, 2018-03-11:
@sakshirawal14 It's working fine. picam runs in the foreground and does not return until you manually stop it. If you want to execute other command while picam is running, run picam in the background by adding `&` at the end of picam command. Please open new issue if you still have issues.