iizukanao/picam #84: Request: add option to specify camera sensor mode

benpurdy, 2017-09-05:
Hello, first of all, thank you for this awesome utility, it's super helpful! I'd love to have the ability to set the specific sensor mode for the camera though, this can make a huge difference in the visual fidelity of the output.

Attached is an example image to illustrate (please ignore the exposure, lighting conditions were very different between the two). The top image is from picam and the bottom image is from a python script I wrote where I explicitly set the sensor mode (mode 2 in this case) via PiCamera.

[image]

I'll go ahead and try to implement this change myself, but the build process seems pretty intense.

benpurdy, 2017-09-06:
Ok, I got the build working (turned out to be pretty easy, just takes a little while) and I have this feature implemented. It makes a huge difference in the video quality for the camera module I'm using. I'll submit a pull request soon.

iizukanao, 2017-09-06:
This is very nice addition.

Actually, I need to update build instructions because recent Raspberry Pi models have enough power to build ffmpeg.

paolohu, 2017-09-06:
I'd love to see updated build instructions to build directly on the Pi - I've been toying with the idea of some enhancements as well but building externally seemed a bit daunting.

benpurdy, 2017-09-06:
I built it on the Pi just by following the instructions starting with the "Build fdk-aac" step and it worked great. Took about an hour from start to finish (building on a Pi 3 using the multi-core make option).

paolohu, 2017-09-07:
cool - thanks for the info - i'll have to give it a shot

paolohu, 2017-09-07:
so question before i start the process - if anyone knows - would it be possible to add a hook to dynamically change the roi and volume settings?

benpurdy, 2017-09-07:
I don't know if the ROI can be changed after the camera is started, you could certainly try it and see what happens though. You could probably take a look at how the ROI is set initially and base the new hook on that as a starting point.

Looking at the code it seems like the audio gain can be changed while the stream is running, if you look around line 4310 in stream.c, there's some code that uses audio_volume_multiply to modulate the volume of the audio samples. Seems like you could just update the value of audio_volume_multiply via a new hook and it should just work? The change would likely be pretty abrupt unless you want to implement some kind of smooth fade.

paolohu, 2017-09-07:
thanks - appreciate the feedback - i was kind of thinking the same thing.  i'll give it a shot and see what happens.