iizukanao/picam #86: I2s Mic not recording

supkm, 2017-09-15:
First of all, thank you for this perfect project.
I have an Issue which I can't solve.
Installed all on a raspberry Zero W. All is working with Camera and a Usb microphone.
The I have to Change to an I2s microphone and the issue began.

I installed and compile all of the modules needed, also with arecord i can record audio and play it after with aplay.
when I start picam:
./picam -w 640 -h 480 --alsadev hw:1,0 -r 44100 --verbose --rtspout -v 500000 -f 20 --autoex

The video is playing fine but the microphone does not record.
I use this to stream direct to the iPad (audio and video together) I can see the video but can't hear anything. When I change to USB mic it works again.

Here is the verbose print from the command: The only thing I can see is the "smaller buffer size"
Can you help me out what can be possible the problem?

video_width=640
video_height=480
video_fps=20.000000
fr_q16=1310720
video_pts_step=4500
video_gop_size=20
video_rotation=0
video_hflip=0
video_vflip=0
video_bitrate=500000
video_avc_profile=constrained_baseline
video_avc_level=3.1
video_qp_min=-1
video_qp_max=-1
video_qp_initial=-1
video_slice_dquant=-1
alsa_dev=hw:1,0
audio_channels=1
audio_sample_rate=44100
audio_bitrate=40000
audio_volume_multiply=1.000000
is_hlsout_enabled=0
is_hls_encryption_enabled=0
hls_encryption_key_uri=stream.key
hls_encryption_key=0x75b0a81de17487c88a47507a7e1fdf73
hls_encryption_iv=0x000102030405060708090a0b0c0d0e0f
hls_output_dir=/run/shm/video
rtsp_enabled=1
rtsp_video_control_path=/tmp/node_rtsp_rtmp_videoControl
rtsp_audio_control_path=/tmp/node_rtsp_rtmp_audioControl
rtsp_video_data_path=/tmp/node_rtsp_rtmp_videoData
rtsp_audio_data_path=/tmp/node_rtsp_rtmp_audioData
tcp_enabled=0
tcp_output_dest=
auto_exposure_enabled=1
auto_exposure_threshold=5.000000
is_vfr_enabled=1
white_balance=auto
exposure_control=auto
awb_red_gain=0.000000
awb_blue_gain=0.000000
metering=average
manual_exposure_compensation=0
exposure_compensation=0.000000
manual_exposure_aperture=0
exposure_aperture=0
manual_exposure_shutter_speed=0
exposure_shutter_speed=0
manual_exposure_sensitivity=0
exposure_sensitivity=0
roi_left=0.000000
roi_top=0.000000
roi_width=1.000000
roi_height=1.000000
min_fps=-1.000000
max_fps=-1.000000
is_timestamp_enabled=0
timestamp_format=%a %b %d %l:%M:%S %p
timestamp_layout=15
timestamp_horizontal_margin=10
timestamp_vertical_margin=10
is_timestamp_abs_pos_enabled=0
timestamp_pos_x=0
timestamp_pos_y=0
timestamp_text_align=1
timestamp_font_name=FreeMono:style=Bold
timestamp_font_file=
timestamp_font_face_index=(null)
timestamp_font_points=14.000000
timestamp_font_dpi=96
timestamp_color=ffffff
timestamp_stroke_color=000000
timestamp_stroke_width=1
timestamp_letter_spacing=0
is_preview_enabled=0
is_previewrect_enabled=0
preview_x=0
preview_y=0
preview_width=0
preview_height=0
preview_opacity=255
blank_background_color=0x0
is_audio_preview_enabled=0
audio_preview_dev=plughw:0,0
record_buffer_keyframes=5
state_dir=state
hooks_dir=hooks
connecting to UNIX domain sockets
configuring devices
using variable frame rate
exposure control: auto
setting exposure:
  eMetering: 0
  xEVCompensation: 0
  nApertureFNumber: 0
  bAutoAperture: 0
  nShutterSpeedMsec: 0
  bAutoShutterSpeed: 1
  nSensitivity: 0
  bAutoSensitivity: 1
Set video_encode state to idle
Enable port buffers for camera 71
Enable port buffers for video_encode 200
Enable port buffers for video_encode 201
Set camera state to executing
Set video_encode state to executing
opening ALSA device for capture: hw:1,0
cannot use mono audio; trying stereo
final audio_channels: 2
sample_rate: 44100
nb_samples: 1024
sample_fmt: 1
audio_codec_ctx->channel_layout: 3
av_frame->channel_layout: 3
audio_codec_ctx->channels: 2
av_frame->channels: 0
allocated 4096 bytes for audio samples
audio_pts_step_base: 2089
actual sample rate=44100 dir=0
failed to set buffer size for microphone: buffer_size=4096 multiply=100
trying smaller buffer size for microphone: buffer_size=4096 multiply=50
failed to set buffer size for microphone: buffer_size=4096 multiply=50
trying smaller buffer size for microphone: buffer_size=4096 multiply=25
microphone: buffer size: 102400 frames (channels=2 buffer_size=4096 multiply=25)
microphone: setting period size to 1024
actual_period_size=1024 dir=0
audio capture device:
Hardware PCM card 1 'snd_rpi_simple_card' device 0 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : MMAP_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 44100
  exact rate   : 44100 (44100/1)
  msbits       : 16
  buffer_size  : 102400
  period_size  : 1024
  period_time  : 23219
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 1024
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 102400
  silence_threshold: 0
  silence_size : 0
  boundary     : 1677721600
  appl_ptr     : 0
  hw_ptr       : 0
start capturing video
waiting for the first video buffer
start capturing audio
[microphone started]capturing started
.[NAL7].....................increase gop_size to 29  31.32 fps k=1 a-v=-873 c-a=-411 u=0 d=0 pts=67721 y=45.4
..increase gop_size to 30  36.01 fps k=2 a-v=-489 c-a=-1281 u=0 d=0 pts=73604 y=50.6
.. 33.44 fps k=3 a-v=-114 c-a=-1139 u=0 d=0 pts=79496 y=47.0
.............................. 30.17 fps k=4 a-v=-585 c-a=-736 u=0 d=0 pts=169794 y=44.2
.............................. 30.28 fps k=5 a-v=-706 c-a=-977 u=0 d=0 pts=259742 y=44.9
.............................. 30.28 fps k=6 a-v=-777 c-a=-866 u=0 d=0 pts=349640 y=44.8
.............................. 30.53 fps k=7 a-v=-721 c-a=-890 u=0 d=0 pts=439411 y=46.7
.............................. 30.28 fps k=8 a-v=-798 c-a=-696 u=0 d=0 pts=529315 y=46.3
.............................. 30.23 fps k=9 a-v=-947 c-a=-180 u=0 d=0 pts=619291 y=46.3
.............................. 30.06 fps k=10 a-v=-877 c-a=232 u=0 d=0 pts=709048 y=46.0
.............................. 30.44 fps k=11 a-v=-1069 c-a=-343 u=0 d=0 pts=799067 y=44.9
.............................. 30.35 fps k=12 a-v=-1163 c-a=-131 u=0 d=0 pts=888988 y=44.6
.............................. 30.35 fps k=13 a-v=-1309 c-a=-1 u=0 d=0 pts=978961 y=44.5
.............................. 30.38 fps k=14 a-v=-1361 c-a=94 u=0 d=0 pts=1068840 y=42.4
.............................. 30.48 fps k=15 a-v=-1527 c-a=329 u=0 d=0 pts=1158833 y=40.5
.............................. 30.39 fps k=16 a-v=-1534 c-a=374 u=0 d=0 pts=1248667 y=35.8
.............................. 30.48 fps k=17 a-v=-1404 c-a=227 u=0 d=0 pts=1338364 y=34.4
.............................. 30.13 fps k=18 a-v=-1496 c-a=370 u=0 d=0 pts=1428283 y=34.0
.............................. 30.13 fps k=19 a-v=-1596 c-a=516 u=0 d=0 pts=1518210 y=33.5
.............................. 30.05 fps k=20 a-v=-1883 c-a=976 u=0 d=0 pts=1608324 y=35.2
.............................. 30.45 fps k=21 a-v=-1784 c-a=800 u=0 d=0 pts=1698052 y=39.9
.............................. 30.13 fps k=22 a-v=-1877 c-a=940 u=0 d=0 pts=1787972 y=31.7
.............................. 30.14 fps k=23 a-v=-1958 c-a=1089 u=0 d=0 pts=1877880 y=35.9
.............................. 30.05 fps k=24 a-v=-2264 c-a=1992 u=0 d=0 pts=1968013 y=36.9
.............................. 30.39 fps k=25 a-v=-2170 c-a=1390 u=0 d=0 pts=2057746 y=38.7
.............................. 30.22 fps k=26 a-v=-2239 c-a=1515 u=0 d=0 pts=2147642 y=38.4
.............................. 30.20 fps k=27 a-v=-2349 c-a=1668 u=0 d=0 pts=2237579 y=38.5
.............................. 30.15 fps k=28 a-v=-403 c-a=-176 u=0 d=0 pts=2327549 y=38.7
.............................. 30.14 fps k=29 a-v=-457 c-a=-121 u=0 d=0 pts=2417430 y=38.6
.............................. 30.06 fps k=30 a-v=-558 c-a=-28 u=0 d=0 pts=2507358 y=38.5
.............................. 30.05 fps k=31 a-v=-613 c-a=140 u=0 d=0 pts=2597240 y=38.5
.............................. 30.13 fps k=32 a-v=-690 c-a=266 u=0 d=0 pts=2687144 y=38.6
.............................. 30.13 fps k=33 a-v=-788 c-a=407 u=0 d=0 pts=2777069 y=41.8
.................^Cstop requested (signal=2)
.Reverting pBuffer to its original value
shutdown sequence start
stop capturing video
shutdown_openmax: ilclient_flush_tunnels
shutdown_openmax: disable port buffer for camera 71
shutdown_openmax: disable port buffer for video_encode 200
shutdown_openmax: disable port buffer for video_encode 201
shutdown_openmax: disable tunnel[0]
shutdown_openmax: teardown tunnels
shutdown_openmax: state transition to idle
shutdown_openmax: state transition to loaded
shutdown_openmax: ilclient_cleanup_components
shutdown_openmax: OMX_Deinit
shutdown_openmax: ilclient_destroy cam_client
shutdown_openmax: ilclient_destroy ilclient
shutdown_video
teardown_audio_encode
teardown_audio_capture_device
hls_destroy
pthread_mutex_destroy
teardown_socks
free_encoded_packets
stop_watching_hooks
stop requested (signal=15)
pthread_join hooks_thread
free keyframe_pointers
shutdown successful

pcunning, 2017-09-21:
Funny - I just ran into the exact same problem! I was setting up the SPH0645LM4H following the direction on Adafruit and ran into the same problem. It turns out the issue is that we didn't know how to address the `pcm.dmic_sv` device. Looking at http://www.alsa-project.org/alsa-doc/alsa-lib/pcm.html lead me to `plug:dmic_sv` and that worked! You should be able to run `./picam -w 640 -h 480 --alsadev plug:dmic_sv -r 44100 --verbose --rtspout -v 500000 -f 20 --autoex` and get audible audio.

supkm, 2017-09-21:
Strange. I will test this fast because my test was "plugdevice:dmic_sv" -> which is not working.

supkm, 2017-09-21:
Hey, it is working. Have no clue what is riding me as I use plugdevice instead of plug.
THANK YOU!

closed 2017-09-21 by supkm