iizukanao/picam #87: libGLESv2 Error

crosofg, 2017-09-15:
I am getting this error on starting?
```
 error while loading shared libraries: libGLESv2.so: cannot open shared object file: No such file or directory
```

xbgmsharp, 2018-01-11:
You are missing a lib.

```
# apt-get install libgles2-mesa-dev
```
You can check:
```
# ls -lh /usr/lib/arm-linux-gnueabihf/libGLESv2*
lrwxrwxrwx 1 root root  18 Aug  4 12:41 /usr/lib/arm-linux-gnueabihf/libGLESv2.so -> libGLESv2.so.2.0.0
lrwxrwxrwx 1 root root  18 Aug  4 12:41 /usr/lib/arm-linux-gnueabihf/libGLESv2.so.2 -> libGLESv2.so.2.0.0
-rw-r--r-- 1 root root 38K Aug  4 12:41 /usr/lib/arm-linux-gnueabihf/libGLESv2.so.2.0.0
```

or 
```
# ldd /usr/local/bin/picam 
	linux-vdso.so.1 (0x7eee4000)
...
```

kllngtme, 2021-05-04:
I'm getting this error on a raspberry pi 4 using buster. I've ensured I've installed libglesv2 as well. Any other ideas on getting this started?

/picam: error while loading shared libraries: libGLESv2.so: cannot open shared object file: No such file or directory