iizukanao/picam #89: Setting the recording time for each file

wikusjacobsz, 2017-10-01:
Hi,

Is there a way you can set the maximum recording time for each file? I want to be able to set recording time for example 1 hour. Once the current file contains an hour of video, the file should be closed and a new file created.

Is this possible, with picam? Or is there another external service that I can use to monitor the file size and use the hooks to stop and start recording?

iizukanao, 2017-10-02:
@wikusjacobsz 
Hi,

picam does not have such built-in feature, but you do it with a shell script like this:

```sh
#!/bin/sh

# Replace ~/picam with your picam directory
cd ~/picam || exit 1

# Stop recording on Ctrl-C
trap on_sigint INT
on_sigint() {
  echo "Stop recording"
  touch hooks/stop_record
  exit 0
}

# Do forever
while :
do
  echo "Start recording"
  touch hooks/start_record || exit 1

  echo "Waiting for one hour..."
  sleep 1h

  echo "Stop recording"
  touch hooks/stop_record || exit 1

  # Wait for recording to stop
  while [ `cat state/record` = "true" ]; do
    sleep 0.1
  done
done
```

wikusjacobsz, 2017-10-06:
@iizukanao Thank you so much for the response. I took your script, modified it slightly and got exactly what I was looking for.

closed 2017-10-06 by wikusjacobsz