iizukanao/picam #90: Creating hook files doesn't seem to perform actions?

kamranayub, 2018-01-03:
I'm writing a little web frontend for Picam called [picam-viewer](https://github.com/kamranayub/picam-viewer) and I got as far as generating files in the hooks directory for Picam.

However, when I create files (even through the shell on the Pi), nothing appears to happen (like unmute/mute) and the files just stay there. Am I missing something? I'm just doing a basic:

    touch ~/picam/hooks/mute

Which is what the docs say, but the stream never mutes (using HLS).

iizukanao, 2018-01-04:
@kamranayub First of all, I appreciate your work and thank you for sharing your code!

When you run picam with `--verbose` option, will the console prints `mute` when you create ~/picam/hooks/mute? Also, the file ~/picam/hooks/mute will automatically be removed immediately after you create the file?

kamranayub, 2018-01-04:
Thanks I will try that, when I tested it, it did not remove the file
immediately. I actually have a recording of me trying it out I think, I
would need to find the timestamp in the video.