iizukanao/picam #92: Recording video+audio LAG

mitsus, 2018-01-05:
Hi all,
I try to record Video+Audio using my RaspiCam with i2S mic. All work fine, but during playback of recorderd video isee somethings some LAG...so I see the video that stop play for 1 sec in various point of it.
I use these basic paramenters for start picam:
`/home/chitchat/picam/picam -w 1920 -h 1080 --alsadev plug:dmic_sv -r 44100`

I use rec directories in RAM (/run/shm). How can I solve these lags.

Thanks and best regards

iizukanao, 2018-01-08:
@mitsus Can you use `plughw:` instead of `plug:`? Also, can you provide full console output of picam with `--verbose` option?

mitsus, 2018-01-12:
Hi @iizukanao below the output with --verbose:

`/home/chitchat/picam/picam -w 1920 -h 1080 -f 25 --alsadev plughw:dmic_sv -r 44100 --verbose
using AVC High Profile Level 4
warn: fps > 20 might not work properly when width and height is large.
      Use lower --fps or use --vfr. If you still want to use this
      configuration, see if picam keeps up with 25.0 fps using --verbose.
video_width=1920
video_height=1080
video_fps=25.000000
fr_q16=1638400
video_pts_step=3600
video_gop_size=25
video_rotation=0
video_hflip=0
video_vflip=0
video_bitrate=2000000
video_avc_profile=high
video_avc_level=4
video_qp_min=-1
video_qp_max=-1
video_qp_initial=-1
video_slice_dquant=-1
alsa_dev=plughw:dmic_sv
audio_channels=1
audio_sample_rate=44100
audio_bitrate=40000
audio_volume_multiply=1.000000
is_hlsout_enabled=0
is_hls_encryption_enabled=0
hls_encryption_key_uri=stream.key
hls_encryption_key=0x75b0a81de17487c88a47507a7e1fdf73
hls_encryption_iv=0x000102030405060708090a0b0c0d0e0f
hls_output_dir=/run/shm/video
rtsp_enabled=0
rtsp_video_control_path=/tmp/node_rtsp_rtmp_videoControl
rtsp_audio_control_path=/tmp/node_rtsp_rtmp_audioControl
rtsp_video_data_path=/tmp/node_rtsp_rtmp_videoData
rtsp_audio_data_path=/tmp/node_rtsp_rtmp_audioData
tcp_enabled=0
tcp_output_dest=
auto_exposure_enabled=0
auto_exposure_threshold=5.000000
is_vfr_enabled=0
white_balance=auto
exposure_control=auto
awb_red_gain=0.000000
awb_blue_gain=0.000000
metering=average
manual_exposure_compensation=0
exposure_compensation=0.000000
manual_exposure_aperture=0
exposure_aperture=0
manual_exposure_shutter_speed=0
exposure_shutter_speed=0
manual_exposure_sensitivity=0
exposure_sensitivity=0
roi_left=0.000000
roi_top=0.000000
roi_width=1.000000
roi_height=1.000000
min_fps=-1.000000
max_fps=-1.000000
is_timestamp_enabled=0
timestamp_format=%a %b %d %l:%M:%S %p
timestamp_layout=15
timestamp_horizontal_margin=10
timestamp_vertical_margin=10
is_timestamp_abs_pos_enabled=0
timestamp_pos_x=0
timestamp_pos_y=0
timestamp_text_align=1
timestamp_font_name=FreeMono:style=Bold
timestamp_font_file=
timestamp_font_face_index=(null)
timestamp_font_points=14.000000
timestamp_font_dpi=96
timestamp_color=ffffff
timestamp_stroke_color=000000
timestamp_stroke_width=1
timestamp_letter_spacing=0
is_preview_enabled=0
is_previewrect_enabled=0
preview_x=0
preview_y=0
preview_width=0
preview_height=0
preview_opacity=255
blank_background_color=0x0
is_audio_preview_enabled=0
audio_preview_dev=plughw:0,0
record_buffer_keyframes=5
state_dir=state
hooks_dir=hooks
configuring devices
exposure control: auto
setting exposure:
  eMetering: 0
  xEVCompensation: 0
  nApertureFNumber: 0
  bAutoAperture: 0
  nShutterSpeedMsec: 0
  bAutoShutterSpeed: 1
  nSensitivity: 0
  bAutoSensitivity: 1
Set video_encode state to idle
Enable port buffers for camera 71
Enable port buffers for video_encode 200
Enable port buffers for video_encode 201
Set camera state to executing
Set video_encode state to executing
opening ALSA device for capture: plughw:dmic_sv
ALSA lib pcm_hw.c:1713:(_snd_pcm_hw_open) Invalid value for card
error: cannot open audio capture device 'plughw:dmic_sv': No such device
hint: specify correct ALSA device with '--alsadev <dev>'
warning: audio capturing is disabled
sample_rate: 44100
nb_samples: 1024
sample_fmt: 1
audio_codec_ctx->channel_layout: 4
av_frame->channel_layout: 4
audio_codec_ctx->channels: 1
av_frame->channels: 0
allocated 2048 bytes for audio samples
audio_pts_step_base: 2089
start capturing video
waiting for the first video buffer
capturing started
.[NAL7].......................... 25.63 fps k=1 a-v=4583 c-a=-1771 u=0 d=0 pts=93600
..........^Cstop requested (signal=2)
Reverting pBuffer to its original value
shutdown sequence start
stop capturing video
shutdown_openmax: ilclient_flush_tunnels
shutdown_openmax: disable port buffer for camera 71
shutdown_openmax: disable port buffer for video_encode 200
shutdown_openmax: disable port buffer for video_encode 201
shutdown_openmax: disable tunnel[0]
shutdown_openmax: teardown tunnels
shutdown_openmax: state transition to idle
shutdown_openmax: state transition to loaded
shutdown_openmax: ilclient_cleanup_components
shutdown_openmax: OMX_Deinit
shutdown_openmax: ilclient_destroy cam_client
shutdown_openmax: ilclient_destroy ilclient
shutdown_video
teardown_audio_encode
hls_destroy
pthread_mutex_destroy
teardown_socks
free_encoded_packets
stop_watching_hooks
pthread_join hooks_thread
stop requested (signal=15)
free keyframe_pointers
shutdown successful`

mitsus, 2018-01-12:
With this options: `/home/chitchat/picam/picam -w 1920 -h 1080 -f 30 --alsadev plug:dmic_sv -r 44100`

all work fine, but video quality is not the best. If i run raspivid with same paramenters it was best of picam.
So How can I have the best quality video with this resolution (or lower resolution)? My purpose is to have best quality video possible.

Thanks a lot