iizukanao/picam #95: Overlaying text (subtitle)

arnauldb, 2018-01-28:
Hello,

I don't understand how to add an overlaying text...Could you help me ?

Thank you for picam, I use it to livestream to youtube and periscope;

iizukanao, 2018-01-30:
@arnauldb Thank you for using picam! All you have to do is create a text file named PICAM_DIR/hooks/subtitle (Replace PICAM_DIR with your picam installation directory) with the contents like `text=Put some text here` while picam is running. picam will automatically read that file and overlay text.

iizukanao, 2018-01-31:
@arnauldb If you want to create a subtitle file with a text editor, create a file named "subtitle" (without extension) in somewhere else (e.g. /home/pi/subtitle) then move the file to the destination (`mv /home/pi/subtitle /home/pi/picam/hooks/subtitle`). Do not add any extension to the filename.

arnauldb, 2018-01-31:
@iizukanao It is working well now, thank you for your help.

nor500, 2018-11-25:
Hello,
I have a problem:
Whenever I try example:
echo 'text=Houston, we have a problem' > hooks/subtitle
I am receiving the following error message:
Fontconfig warning: ignoring UTF-8: not a valid region tag